sportiduino/sportiduino #1: Сделать перевернутый режим работы - у спортсменов станции, на кп метки

alexandervolikov, 2017-10-12:


closed 2018-01-08 by alexandervolikov

Evgen-F, 2020-09-03:
Почему отказались от данной идеи?

alexandervolikov, 2020-09-03:
> Почему отказались от данной идеи?

Получается не очень удобно в современных реалях. Станции довольно массивные для ношения.

Evgen-F, 2020-09-03:
> > Почему отказались от данной идеи?
> 
> Получается не очень удобно в современных реалях. Станции довольно массивные для ношения.

Тем не менее в определенных случаях оно имеет смысл. Мне например на тренировках в черте города значительно спокойнее повесить метки NTAG/MiFare на контрольные пункты - они нафиг никому не сдались, а в качестве чипов использовать станции - не такие они и большие выходят. Зато вероятность убытков от кражи станции стремится к нулю, и появляется возможность использовать эту систему электронной отметки на каждой тренировке.

mforis, 2020-09-03:
Оно то может и так, но участник может упасть и разбить базу, пусть вероятность мала, но так может быть. Да и от вибрации может что-нибудь повредиться. Я думаю, стоило бы подумать над бесконтактными базами, например, на финиш, чтобы участник просто пробегал через финишную арку и отметка уже в чипе.