sportiduino/sportiduino #2: Сделать более легкособираемый вариант станции

alexandervolikov, 2017-10-12:
На материнской плате собирается навесным монтажом из блоков: Arduino pro mini, Ds3231, RC522.
Большая герметичная коробка, например, G203CMF

burunduk, 2017-11-21:
Про вариант: У китайцев популярна плата часов с батарейкой CR2032 и штырями.
это не увеличит автономность?

alexandervolikov, 2017-11-22:
Да, причем в плате с часами и батарейкой есть уже епром на 4к. Её и буду использовать для данного варианта. Платы для сборки из модулей уже приехали, скоро соберу, выложу что получилось.

alexandervolikov, 2019-01-11:
Поэкспериментировав, решил отказаться от этой идеи. С учетом необходимых доработок готовых плат (удаление лишних диодов и пр.), большего объема пайки (каждую ножку платы), проще и быстрее использоваться отдельные микросхемы как это реализовано сейчас в основном варианте.

closed 2019-01-11 by alexandervolikov