sportiduino/sportiduino #5: Чипы на запястье или на палец

alexandervolikov, 2017-10-12:
Заказ и тестирование.

alexandervolikov, 2017-10-24:
Возможность есть. Заказал и протестировал несколько образцов.

Чипы в виде браслета (но можно и на палец крепить резинкой) стоят из Китая при заказе партии в 1000 штук - по 20 рублей, 500 штук - 30 рублей. Но сюда ещё нужно включить стоимость доставки, возможные проблемы с растаможкой. 
Искать на алибабе по "rfid wristband". Нужный чип - F1108 Fudan. Это полный Китайский дешевый аналог Mifare Classik 1K. Под дешевыми чипами подразумеваются именно они.

В России при беглом поиске и переписке с поставщиками нашёл чипы по цене от 70 рублей за чип в виде браслета с текстильным ремешком.

closed 2017-10-24 by alexandervolikov

mforis, 2017-10-24:
А подходят ли чипы которые идут в комплекте с rfid считывателем (белые карточки и синие чипы) к этому проекту?

alexandervolikov, 2017-10-24:
Да, подходят. Там FM1108, китайский аналог Mifare Classik 1K