sportiduino/sportiduino #9: Заливка прошивки в Atmega.

mforis, 2017-10-20:


mforis, 2017-10-20:
Можно ли залить прошивку через Arduino Uno без программатора?

alexandervolikov, 2017-10-20:
Наверное, как-то это можно осуществить, если сделать прошивку совмещенную с буатлоадером, который умеет обрабатывать корректно прерывания от watchdog. Если заливать как есть просто через ArduinoAsISP, то она не работает, уходит в bootloop.

mforis, 2017-10-20:
Блииииннн... Очень плохо... Необходимо где-то срочно искать программатор...

mforis, 2017-10-20:
А что у вас за программатор?

alexandervolikov, 2017-10-20:
Я пользуюсь самым дешевым USB TTL-UART на чипе CHG340
https://ru.aliexpress.com/store/product/CH340G-5V-USB-to-TTL-UART-Serial-Converter/1950989_32524218818.html

Сейчас глянул, да можно обойтись и Arduino UNO для заливки прошивки, но из UNO нужно вытащить чип Atmega, если это возможно и замкнуть Reset на землю (https://forum.arduino.cc/index.php?topic=146258.0)
Тогда Arduino UNO превраиться в USB to UART и можно залить прошивку.

Но перед этим всё-равно надо залить правильный буатлодер - Optiboot
(https://github.com/Optiboot/optiboot)
пользуясь ArduinoAsISP на UNO.

mforis, 2017-10-20:
Александр, еще вопрос, а если у меня копия Arduino Uno с чипом CH340G? Могу ли я сразу прошить или тоже необходимо выполнять указания выше. Спасибо и извините за назойливость)))

alexandervolikov, 2017-10-20:
Сначала  залить прошивку ArduinoAsISP на Arduino UNO
(File - Examlpes - ArduinoAsISP). На Arduino UNO разместить конденсатор 10 мкФ между RST и GND. Подключить к программируемой плате пины 11,12,13 к MOSI, MISO, SCK соответственно. Пин 10 подключить к RST программируемой платы и запитать плату.

Далее установить Optiboot в IDE:
вставить URL-адресс:
https://github.com/Optiboot/optiboot/releases/download/v6.2/package_optiboot_optiboot-additional_index.json
в  "Additional Boards Manager URLs" в File-Prefernces
После этого прошивки станут ддоступны через меню Tools-Board
Для прошивки  нужно выбрать:
Board: Optiboot on 32 pins
Processor: ATmega328p
CPU Speed: 8Mhz (int)
Programmer: Arduino As ISP

Нажать Burn bootloader
Если всё нормально, появиться надпись 
Done burning boatloader 
Если нет, то где допущена ошибка, проверить правильность подключения, пайки и т.п.

Дальше отключаем UNO от программируемой платы. В IDE меняем программатор на "AVRISP mkII". Замыкаем на UNO пины  RST и GND. Подключаемм UNO к программируемой плате: RX-TX, TX-RX. Вроде как, если использовать UNO в качестве USB-UART, то пин RST на программируемой плате придётся нажимать (заземление для перезагрузки платы) вручную. Говорят, это нужно сделать в нужный момент во время загрузки прошивки, который определяется опытным путем.
С отдельным USB-UART, в котором есть DTR проще. Стоит он недорого, в том числе и в России.

alexandervolikov, 2017-10-20:
Но, если хочется залить прошивку просто для теста. То можно из Stantion.ino удалить строчки:
sleep(work);
//Power down regime: sleep 24 s
  if (deepsleep) {
    for (uint8_t i = 0; i<3; i++){
      sleep8s();
    }
  }
Это отключит вход в энергосберегающие режимы, будет кушать значительно больше батарейку, но проверить функциональность можно. И залить через ArduinoAsISP.

mforis, 2017-10-20:
Спасибо ОГРОМНОЕ! Если не сложно закиньте эту инструкцию в Wiki.

closed 2017-10-20 by alexandervolikov

mforis, 2017-10-22:
Нашел у себя Arduino Nano с чипом CH340G, как с помощью неё прошить Atmega? По той же инструкции или как-то по другому?

alexandervolikov, 2017-10-22:
Да, с Arduino Nano всё тоже самое. Только нужно ещё нужно установить драйвера на Windows для схемы CHG340, чтобы Windows видела эти устройства.
Да, с платой версии 0.41 всё будет работать. Сейчас и в прошивке по-умолчанию внешняя eeprom выключена, за включение/выключение отвечает строчка:
const boolean extEeprom = false;

mforis, 2017-11-19:
Александр, пришли мне Atmegи и я впаял, начал заливать бутлоадер, но он не прошивается с такой ошибкой. Как решить проблему???
[image]

alexandervolikov, 2017-11-19:
Вероятно, схема плохо пропаялась. Проверьте с лупой или тестером все контакты.

mforis, 2017-11-19:
Блин, я всё уже по пять раз проверил, даже вторую плату собрал, но такая-же фигня...

alexandervolikov, 2017-11-19:
А конденсатор 10 мкФ на uno между rst и gnd стоит?

mforis, 2017-11-19:
Да, сейчас взял Нано, и выбило другую ошибку...
[image]

alexandervolikov, 2017-11-19:
У меня такая ошибка, обычно, выходит когда я плохо припаял. Казалось бы, всё делал четко и последовательно, все контакты пропаяны, но ошибка. Тогда ещё раз проходился паяльников по контактам и оно начинало работать. Худший вариант что может быть не так: атмега сгорела, например, если передержать паяльник на контактах.

Можете сделать следующее: снять с uno (или nano) конденсатор. Загрузить вот эту прошивку:
https://yadi.sk/d/aYVpiodu3PqAwH
И открыть Serial-port в IDE Arduino при подключенной программируемой плате. Это прошивка проводит диагностику и должна выдать то, что видит на SPI контактах.

mforis, 2017-11-19:
[image]

mforis, 2017-11-19:
[image]

mforis, 2017-11-19:
Капец как я расстроился, нифига не получается, ничего не работает! И не могу понять в чем косяк.
Пайку проверил, пропоял еще раз и ничего...

alexandervolikov, 2017-11-19:
Да, соединения с микроконтроллером нет. А как Вы соединялись платы? Можете прислать фото?
Не расстраивайтесь, я тоже бывало по нескольку часов сидел над платами в поисках ошибки

mforis, 2017-11-19:
Да, сейчас скину фотки....

mforis, 2017-11-19:
[image]
[image]
[image]

mforis, 2017-11-19:
[image]
[image]

alexandervolikov, 2017-11-19:
У Вас rst контроллера не подключен, это тот который уходит в другой конец платы (там где два ряда по три дырки, сейчас он у Вас не распаян). Тот rst, который подключен к Вас относится к rc522 и при прошивке он не задействован. 
Если не получится, еще можно попробовать прошить при 5в (переставить в uno соответствующий штырек )

alexandervolikov, 2017-11-19:
Вот сравните с моими фото при прошивке
https://user-images.githubusercontent.com/32615135/32993556-0f472c92-cd6b-11e7-958f-a5bdd66cdeda.jpg
https://raw.githubusercontent.com/alexandervolikov/sportiduino/master/Images/s11.jpg

mforis, 2017-11-19:
Всё бут прошил!
[image]

mforis, 2017-11-19:
Теперь проблема с заливкой скетча, когда нажимать ресет и где? На ардруине?
[image]
[image]
[image]

alexandervolikov, 2017-11-19:
Нет, нажимать нужно на програмируемой плате. Для этого в нужный момент нужно заземлить и отпустить контакт rst (тот же который в двух рядах по три дырки). Сам таким путем никогда не прошивал, поэтому тонкостей не знаю, но читал, что момент нажатия подбирается опытным путем, нужно некоторое число попыток.

alexandervolikov, 2017-11-19:
Момент засеките по прогрессу загрузке в окне ардуино внизу справа, должно быть около 80% заполнения этой полосы. Когда скетч уже откомпилируется

mforis, 2017-11-19:
Хорошо, спасибо. Завтра буду продолжать программирование плат.

mforis, 2017-11-20:
После нажатия заземление RST процесс зависает....
[image]

mforis, 2017-11-20:
[image]

alexandervolikov, 2017-11-20:
Попробуйте сделать вот по этой инструкции:
https://www.youtube.com/watch?v=jk-uq3dlvKs
Только выбрав нужную прошивку и программатор.

alexandervolikov, 2017-11-20:
То есть, задача такая же как прошить Arduino Pro Mini. В сети есть инструкции.

mforis, 2017-11-20:
Пока залить скетч не получается....😢

alexandervolikov, 2017-11-20:
Есть так же вероятность, что Rx Tx плохо пропаяны. Ещё также важно подключение попарно Rx-Tx, Tx-Rx. Возможно стоит питание подвести от 5 вольт при прошивке (у меня где-то платы 3 из 50 прошиваются только таким образом)

Но, возможно, что просто не удается поймать момент.
Тогда два пути. Заказать вот такую штуку:
https://ru.aliexpress.com/store/product/CH340G-5V-USB-to-TTL-UART-Serial-Converter/1950989_32524218818.html
Можно и из России, стоят они не дорого и там есть контакт Dtr, который собственно нажимает кнопку reset в нужный момент. (в данном экземпляре неправильная маркировка и соединять нужно Rx-Rx, Tx-Tx).

Ещё можно пожертвовать какой-нибудь ардуиной с сериал-портом (нано или уно), выпаять от туда Atmegу и использовать как в видео, которое я прилагал выше.

alexandervolikov, 2017-11-20:
Да, кстати, сейчас посмотрел фотографии, что Вы присылали выше. На тех фотографиях у Вас подключен неправильный Reset. Выше я уже писал, что при прошивке микроконтроллера нужно использовать контакт, который находится в разъемах два ряда по три дырки. Reset, который находится ближе к микроконтроллеру относится к плате RC522.

mforis, 2017-11-20:
А может проблема этом?
[image]

alexandervolikov, 2017-11-20:
Не думаю. У меня при прошивке буатлоадера пишет тоже самое.

mforis, 2017-11-20:
Ясно. Буду капать дальше...

mforis, 2017-11-20:
А в плате RX слева и TX справа? А то шелкография не очень понятна...

alexandervolikov, 2017-11-20:
Rx - ближний к микросхеме контакт, Tx - дальний

mforis, 2017-11-20:
Самое интересное, что бут шьётся без "Б", а скетчи не хотят...

mforis, 2017-11-20:
Уже и МК выпаивал... И чего только не делал...

alexandervolikov, 2017-11-20:
А пришлите фото как Вы с выпаянной МК подключаете платы между собой

mforis, 2017-11-20:
Заработало!!!!!!!!!!!!!!!!!!!

mforis, 2017-11-20:
Прошивал по этой инструкции, только с конденсатором. 
https://youtu.be/Rz1tFghOoiQ

alexandervolikov, 2017-11-20:
Поздравляю! :-)

mforis, 2017-11-20:
И опять новая проблема.... База запускается пикает три раза, через какое-то время пикает еще раз... И  всё больше от неё нет реакции...

alexandervolikov, 2017-11-20:
Предварительно стоит записать мастер-чипы номера станции (240 например), обновления времени и пару обычных чипов. После вставления батареек станция три раза пикнет, что говорит и том, что часы выставлены не верно. Затем через 5 секунд будет длинный сигнал, сигнализирующий, что программа вошла в цикл работы. По умолчанию это режим сна, поэтому станция среагирует на чип устаноки номера станции только через 25 секунд после её поднесения, но можно сразу положить станцию на чип перед вставкой батарей. Станция запишет новый номер ,перезагрузится, и теперь уже способна реагировать на обычные чипы. Подносим чип, ждем до 25 секунд, станция переходит в рабочий режим. Но часы по-прежнему сбиты. В рабочем цикле подносим мастер-чип времени и после перезагрузки подносим второй обычный чип. Считываем его на станции сопряжения и убеждаемся в корректной работе. Если всё так, то станцию можно отмывать. Если нет, то нужно поискать возможные ошибки, прежде всего, в спайке модуля RFID.

alexandervolikov, 2017-11-21:
Принципиальную работу модуля ещё можно проверить, если залить туда прошивку из примера DumpInfo библиотеки RC522:
https://github.com/alexandervolikov/sportiduino/blob/master/Libraries/RFID/examples/DumpInfo/DumpInfo.ino

mforis, 2017-11-21:
При подносе мастер чипа база 6 раз пикает, потом пикает  быстро 3 раза и начинает бысро пикать в чем проблема?

alexandervolikov, 2017-11-21:
Возможно, бутлоадер не тот. Вы заливали при прошивке optibut 32pin 8mhz?

mforis, 2017-11-21:
[image]

mforis, 2017-11-21:
А кстати, вы есть в ВК? Может там будет удобнее общаться?

alexandervolikov, 2017-11-21:
Да, вроде всё верно. Просто признаки - беспрерывное пикание может указывать на то, что вотчдог работает некорректно. Не знаю в чём проблема. А можете скинуть фото спаянной схемы?
В вк я есть, но там общаться не очень удобно. Лучше здесь или по почте:
ab.volikov@gmail.com

mforis, 2017-11-21:
А может это следствие моего метода прошивки?

alexandervolikov, 2017-11-21:
Не, не думаю.
А какую версию прошивки Вы загружали?

mforis, 2017-11-21:
0.43

[4 more replies]