sportiduino/sportiduino #13: Баг с записью чипов

alexandervolikov, 2017-11-13:
С некоторым шансом при плохих условиях прикладывания чипа станция пишет отметку не в последнюю пустую страницу а в какую-то промежуточную.
Вероятно, при считывании с какой-то пустой страницы, на станцию приходит, что она не пустая.
Из-за этого явления возникают критические ошибки с перезаписью уже существующих данных на страницы.
Возможно, при очистке нужно писать в страницы не нули, а какую-то контрольную информацию.

mforis, 2017-11-14:
Скиньте пожалуйста, самую стабильную на данный момент прошивку.

alexandervolikov, 2017-11-14:
Точно стабильная - это 0.37. На ней проводились соревнования, и всё было хорошо. Но она рассчитана на 46 отметок
https://github.com/alexandervolikov/sportiduino/tree/fbe66d3bbb6a0b0fddf518feaf086d0d8c350f17
Если важно именно 92 отметки, то, вероятно, последняя устойчивая версия - 0.41. До введения бинарного поиска.
https://github.com/alexandervolikov/sportiduino/tree/7e51d193c366d96c4dc2d69648cfd7573fe4ffbb

mforis, 2017-11-14:
Спасибо, пока достаточно 46 отметок. Со дня-на-день придут МК и до соберу и буду прошивать. Уже не терпится провести полевые испытания))) фотки:
[image]
[image]

alexandervolikov, 2017-11-15:
Отлично! Если что, пишите. А баг я постараюсь в скором времени ликвидировать.

alexandervolikov, 2017-11-19:
Баг исправил. Проблема возникла из-за того, что функция поиска последней страницы в случае неудачного поиска (если чип был приложен плохо) не вызывала выход из функции записи чипов, в результате чего станция могла записать метку в страницу, которую считала у предыдущего чипа.
Сейчас исправлено. Также решил отказаться от 92 меток и записывать на Mifare Classic только 46 меток, по числу страниц, чтобы повысить надежность записи.

closed 2017-11-19 by alexandervolikov