sportiduino/sportiduino #15: Вопросы по работе Smanager.

mforis, 2017-12-10:
1. Допустим мне необходимо "усыпить" 10 баз, мне каждый раз необходимо перезаписывать чип "sleep"?
2. Какие действия мне необходимо провести для проведения соревнований в "в заданном направлении" (в Wiki не очень понятно описано)?
3. Можно ли как-то вывести на печать таблицу сплитов?

Работаю с версией 0.43.1

alexandervolikov, 2017-12-10:
1. Да, каждый раз нужно перезаписывать. Сделано так, чтобы предотвратить возможную ошибку, когда мастер-чип случайно попадает в руки участника, и он просто бежит по дистанции и выключает все станции.
Если хочется, чтобы было без перезаписи, можете стереть строки 972-975 в прошивке станций.

2. Составить протокол (protocol.csv), составить дистанции заданного направления (permc.csv) там в первом столбце дистанции, далее по строкам идёт порядок взятия кп, исключая стартовую станцию). Ну а дальше проводить считывание результатов, считать один или все чипы, нажать Load, включить режим заданного направления - кнопка PermC (подсветится синим), выбрать нужные группы, нажать Calc для расчёта и Save для сохранения сплитов. Программа отсортирует по времени на дистанции и по соблюдению заданного направления.
Основательно этот режим я не тестировал, могут выплыть какие-нибудь баги.

3. Нет, нельзя. Smanager развивать и добавлять принципиально новый функционал не планирую. В программе, которую буду разрабатывать взамен, попробую эту функцию включить. Но, это, скорее, где-то только к лету.

mforis, 2017-12-11:
А в таблице что нибудь должно отображаться?
[image]

alexandervolikov, 2017-12-11:
Да, должно. И этих полос быть не должно. Видимо, что-то пошло не так. Пришлите архив /data, посмотрю.

mforis, 2017-12-11:
Да, сейчас пришлю...

mforis, 2017-12-11:
[data.zip](https://github.com/alexandervolikov/sportiduino/files/1547825/data.zip)

alexandervolikov, 2017-12-11:
[Smanager.zip](https://github.com/alexandervolikov/sportiduino/files/1547849/Smanager.zip)
Вот, теперь вроде работает.
Перевёл все текстовые данные в файлах protocol.csv и permc.csv в кодировку UTF-8
(допишу в wiki, что всё работает только в этой кодировке текста)
Ещё изменил саму прогу - переменной specetap в 15 строчке выставил false.

alexandervolikov, 2017-12-11:
Только, похоже, проверка заданного направления работает некорректно. Позже попробую поправить.

mforis, 2017-12-11:
Хорошо, буду ждать, а то вручную проверять - не очень удобно)))

alexandervolikov, 2017-12-13:
Посмотрел, поправил немного программу:
[Smanager.zip](https://github.com/alexandervolikov/sportiduino/files/1554023/Smanager.zip)
Прописывать в дистанцию заданки нужно все кп кроме старта, то есть: 31,245 в данном случае. Смотрите редактированные файл permc.csv
Ещё у Вас на станциях сбито время, поправил его вручную, чтобы корректно считало.

mforis, 2017-12-16:
Такой вопрос, могу ли я использовать одну базу в качестве базы и финиша

alexandervolikov, 2017-12-16:
> Такой вопрос, могу ли я использовать одну базу в качестве базы и финиша
Да, можно. Программа ведёт рассчёт исходя из того, что первая отметка в чипе - старт, последняя - финиш.

Но нужно соответствующим образом настроить сами станции. Сейчас в них по-умолчанию настроено так, что используются отдельные станции старта (номер 240) и отдельные станции финиша (номер 245). Байт настроек редактируется в файле settinfs.csv и передается вместе с паролем на станцию.
После вхождения в режим сна и последующего просыпания станция сбрасывает все настройки и пароли на значения по-умолчанию, и, если хочется избавиться от этой функции совсем, нужно немного редактировать прошивку станций.

alexandervolikov, 2018-01-08:
Поддержка программы завершена. 
В планах написание модуля для программы SportOrg.

closed 2018-01-08 by alexandervolikov