sportiduino/sportiduino #21: Пакет переменной длины

sembruk, 2018-02-16:
Предлагаю сделать пакет переменной длины вместо фиксированной (32 байта).

sembruk, 2018-02-16:
Например,
`0xFE FE FE FE 4B 00 4B`
 — запрос на считывание чипа
вместо
`0xFE FE FE FE 4B 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 4B`

alexandervolikov, 2018-02-16:
Да, можно. Но, чтобы не переполнить память ардуино, лучше тогда функцию и длину пакета поставить в конец передачи. 

Предлагаю такой вариант:

начало передачи (FE FE FE FE), данные, длина данных, функция, контрольная сумма, конец передачи (FD FD FD FD),

например,

запрос на считывание:
`FE FE FE FE 00 4B 4B FD FD FD FD`

ответ станции на считывание чипа с одной отметкой:
`FE FE FE FE 00 23 00 00 00 00 00 00 00 00 10 5A 86 EF 64 15 63 DE FD FD FD FD`

00 23 - номер чипа, 35
00 00 00 00 00 00 00 00 - информация из 6 и 7 страницы чипа, пусто
10 - номер станции, 10
5A 86 EF 64 - unix время, 1518792548
15 - длина данных
63 - функция
DE - контрольная сумма

sembruk, 2018-02-16:
Не понял, зачем длину переносить в конец.
Максимальную длину пакета можно оставить 32. Контрольная сумма в конце.
Вместо четырёх 0xFE в начале можно оставить один. Вполне рабочий вариант.
При разборе пакета читаем длину. Если она больше 25, читаем 25 байт данных и после контрольную сумму.
Если байт длины в процессе передачи испортился, то контольная сумма не сойдётся и ошибки не будет.

alexandervolikov, 2018-02-16:
Хорошо. Давай тогда максимальная длина пакета будет фиксированная и равна 32 байтам. Тогда, по предложенной тобой структуре:

если пакет единичный или последний:
старт (FE), функция, длина пакета, данные, контрольная сумма
если пакет не последний:
старт (FE), функция, номер пакета + 1E, данные, контрольная сумма

то есть, в вышеобозначенных примерах,
запрос чтения:
`FE 4B 00 4B`

ответ:
`FE 63 15 00 23 00 00 00 00 00 00 00 00 10 5A 86 EF 64 DE `

commit c06ff9f6 (2018-02-17): Fixes #21
closed 2018-02-17 by alexandervolikov