sportiduino/sportiduino #24: Проблема с NTAG213

mforis, 2018-03-03:
Купил я себе чипы NTAG213 в виде наклеек, прошил базу чтения на самую новую прошивку, скачал самый новый софт, но при попытке инициализировать чип, база подает три последовательных сигнала, пробовал несколько разных наклеек, но результат тщетен. В чем может быть проблема?

sembruk, 2018-03-03:
Можно попробовать с прошивкой v1.1 и соответсвующим этой версии софтом.

alexandervolikov, 2018-03-03:
Последние прошивки только что ещё раз проверил. Всё работает.
Тут, вероятно, присутствуют аппаратные проблемы. Три сигнала станция подает, когда передача сигнала некорректная, например, если связь между чипом и станцией нестабильная.  Ntag213 отличается от Mifare1K тем, что требуют большую энергию при передаче, поэтому целесообразно в модуле RC522 перепаять индуктивность на более мощную. Хотя в виде наклеек должны бы работать и без перепайки.
А что пишет в лог при поднесении чипа и нажатии кнопки "Read raw"?

closed 2018-03-12 by alexandervolikov