sportiduino/sportiduino #25: Ошибка при прошивке.

mforis, 2018-03-07:
При попытке залить прошивку версии 0.43.1 выбивает вот такую ошибку: Arduino: 1.8.5 (Windows 7), Плата:"Optiboot on 32-pin cpus, 8MHz (int), ATmega328p"

C:\Users\РњРђРљРЎР?Рњ Р¤РћР Р?РЎР¬\Desktop\sportiduino-master Р’РµСЂСЃРёСЏ 0.43\Mark stantion\Arduino_Scetches\Stantion\Stantion.ino:20:24: warning: multi-line comment [-Wcomment]

In file included from C:\Users\РњРђРљРЎР?Рњ Р¤РћР Р?РЎР¬\Desktop\sportiduino-master Р’РµСЂСЃРёСЏ 0.43\Mark stantion\Arduino_Scetches\Stantion\Stantion.ino:7:0:

C:\Program Files (x86)\Arduino\libraries\RFID/MFRC522.h:408:10: warning: extra qualification 'MFRC522::' on member 'MIFARE_Ultralight_Key' [-fpermissive]

     void MFRC522::MIFARE_Ultralight_Key (byte *arr);

          ^

In file included from C:\Users\РњРђРљРЎР?Рњ Р¤РћР Р?РЎР¬\Desktop\sportiduino-master Р’РµСЂСЃРёСЏ 0.43\Mark stantion\Arduino_Scetches\Stantion\Stantion.ino:8:0:

C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\EEPROM\src/EEPROM.h:43:30: warning: type qualifiers ignored on function return type [-Wignored-qualifiers]

     operator const uint8_t() const       { return **this; }

                              ^

C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\EEPROM\src/EEPROM.h:92:26: warning: type qualifiers ignored on function return type [-Wignored-qualifiers]

     operator const int() const          { return index; }

                          ^

C:\Users\РњРђРљРЎР?Рњ Р¤РћР Р?РЎР¬\Desktop\sportiduino-master Р’РµСЂСЃРёСЏ 0.43\Mark stantion\Arduino_Scetches\Stantion\Stantion.ino: In function 'void setup()':

Stantion:88: error: 'struct ts' has no member named 'unixtime'

C:\Users\РњРђРљРЎР?Рњ Р¤РћР Р?РЎР¬\Desktop\sportiduino-master Р’РµСЂСЃРёСЏ 0.43\Mark stantion\Arduino_Scetches\Stantion\Stantion.ino: In function 'void i2cEepromWriteTime(unsigned int)':

Stantion:301: error: 'struct ts' has no member named 'unixtime'

C:\Users\РњРђРљРЎР?Рњ Р¤РћР Р?РЎР¬\Desktop\sportiduino-master Р’РµСЂСЃРёСЏ 0.43\Mark stantion\Arduino_Scetches\Stantion\Stantion.ino: In function 'uint32_t i2cEepromReadTime(unsigned int)':

C:\Users\РњРђРљРЎР?Рњ Р¤РћР Р?РЎР¬\Desktop\sportiduino-master Р’РµСЂСЃРёСЏ 0.43\Mark stantion\Arduino_Scetches\Stantion\Stantion.ino:340:41: warning: ISO C++ says that these are ambiguous, even though the worst conversion for the first is better than the worst conversion for the second:

In file included from C:\Users\РњРђРљРЎР?Рњ Р¤РћР Р?РЎР¬\Desktop\sportiduino-master Р’РµСЂСЃРёСЏ 0.43\Mark stantion\Arduino_Scetches\Stantion\Stantion.ino:4:0:

C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\Wire\src/Wire.h:64:13: note: candidate 1: uint8_t TwoWire::requestFrom(int, int)

     uint8_t requestFrom(int, int);

             ^

C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\Wire\src/Wire.h:61:13: note: candidate 2: uint8_t TwoWire::requestFrom(uint8_t, uint8_t)

     uint8_t requestFrom(uint8_t, uint8_t);

             ^

C:\Users\РњРђРљРЎР?Рњ Р¤РћР Р?РЎР¬\Desktop\sportiduino-master Р’РµСЂСЃРёСЏ 0.43\Mark stantion\Arduino_Scetches\Stantion\Stantion.ino: In function 'uint8_t coun(uint8_t, uint8_t)':

C:\Users\РњРђРљРЎР?Рњ Р¤РћР Р?РЎР¬\Desktop\sportiduino-master Р’РµСЂСЃРёСЏ 0.43\Mark stantion\Arduino_Scetches\Stantion\Stantion.ino:667:9: warning: statement has no effect [-Wunused-value]

C:\Users\РњРђРљРЎР?Рњ Р¤РћР Р?РЎР¬\Desktop\sportiduino-master Р’РµСЂСЃРёСЏ 0.43\Mark stantion\Arduino_Scetches\Stantion\Stantion.ino: In function 'void checkTime()':

Stantion:700: error: 'struct ts' has no member named 'unixtime'

Stantion:700: error: 'struct ts' has no member named 'unixtime'

C:\Users\РњРђРљРЎР?Рњ Р¤РћР Р?РЎР¬\Desktop\sportiduino-master Р’РµСЂСЃРёСЏ 0.43\Mark stantion\Arduino_Scetches\Stantion\Stantion.ino: In function 'void rfid()':

Stantion:866: error: 'struct ts' has no member named 'unixtime'

C:\Users\РњРђРљРЎР?Рњ Р¤РћР Р?РЎР¬\Desktop\sportiduino-master Р’РµСЂСЃРёСЏ 0.43\Mark stantion\Arduino_Scetches\Stantion\Stantion.ino:724:12: warning: unused variable 'code' [-Wunused-variable]

C:\Users\РњРђРљРЎР?Рњ Р¤РћР Р?РЎР¬\Desktop\sportiduino-master Р’РµСЂСЃРёСЏ 0.43\Mark stantion\Arduino_Scetches\Stantion\Stantion.ino: In function 'bool writeTime(int)':

Stantion:1060: error: 'struct ts' has no member named 'unixtime'

Stantion:1069: error: 'struct ts' has no member named 'unixtime'

C:\Users\РњРђРљРЎР?Рњ Р¤РћР Р?РЎР¬\Desktop\sportiduino-master Р’РµСЂСЃРёСЏ 0.43\Mark stantion\Arduino_Scetches\Stantion\Stantion.ino:1050:8: warning: unused variable 'ok' [-Wunused-variable]

C:\Users\РњРђРљРЎР?Рњ Р¤РћР Р?РЎР¬\Desktop\sportiduino-master Р’РµСЂСЃРёСЏ 0.43\Mark stantion\Arduino_Scetches\Stantion\Stantion.ino:1051:11: warning: unused variable 'syndrome' [-Wunused-variable]

C:\Users\РњРђРљРЎР?Рњ Р¤РћР Р?РЎР¬\Desktop\sportiduino-master Р’РµСЂСЃРёСЏ 0.43\Mark stantion\Arduino_Scetches\Stantion\Stantion.ino:1052:12: warning: unused variable 'decode' [-Wunused-variable]

exit status 1
'struct ts' has no member named 'unixtime'
[image]

alexandervolikov, 2018-03-07:
Похоже, что не установлена библиотека ds3231.
Попробуйте скопировать библиотеку https://github.com/alexandervolikov/sportiduino/tree/master/Libraries/ds3231-master
в папку где они у Вас хранятся, например в
C:\Users\username\Documents\Arduino\libraries

alexandervolikov, 2018-03-07:
Похоже, что вообще все библиотеки слетели и из нужно заново подключить.
Библиотеку Хамминга я удалил из репоизтория, прилагаю. 
[Hamming.zip](https://github.com/alexandervolikov/sportiduino/files/1789495/Hamming.zip)

alexandervolikov, 2018-03-07:
Добавил поддержку чипов Mifare S50. Написал отдельные прошивки для работы с ними. Для мастер станции и базовых станций соответственно. 

Работают чуть медленнее. Число отметок до 42. Но полностью поддерживается разрабатываемым программным обеспечением. Потом ещё внесу это в основной репозиторий. А пока просто прилагаю тут. Лучше пользоваться данными версиями чем 0.43.1.

[SportiduinoMifare.zip](https://github.com/alexandervolikov/sportiduino/files/1790580/SportiduinoMifare.zip)

sembruk, 2018-03-07:
Есть смысл оформить это в виде ветки. Могу сделать.

alexandervolikov, 2018-03-07:
Да, давай, если не затруднит.

sembruk, 2018-03-07:
Вот
https://github.com/alexandervolikov/sportiduino/tree/mifare_s50

mforis, 2018-03-08:
Ооо, круто, сегодня прошью свои базы на новую прошивку)))

mforis, 2018-03-08:
Как залить прошивку без выпаивания RFID-модуля?

alexandervolikov, 2018-03-08:
Точно так же. Подключить rx, tx, rst usb-uart, запитать, прошить. Модуль rc522 не влияет на этот процесс, его выпаивать не надо.

mforis, 2018-03-08:
Такая проблема, прошил я значит базы на новую прошивку. Но при поднесении любого мастер чипа станция издает три очень быстрых сигнала. В чем проблема?

mforis, 2018-03-08:
Очень срочно нужна помощь!!!

alexandervolikov, 2018-03-08:
Возможно, дело в установленных паролях мастер чипов. Нужно выставить 0, как по умолчанию после прошивки. Но не уверен. Сейчас возможности проверить и протестировать нет, только после праздников.

mforis, 2018-03-08:
//master chips number in the first three bytes of the chip in the first page
const uint8_t timeMaster = 250;
const uint8_t numberMaster = 251;
const uint8_t sleepMaster = 252;
const uint8_t dumpMaster = 253;
const uint8_t passMaster = 254;

Что означают эти строки?

sembruk, 2018-03-08:
Коды мастер чипов
https://github.com/alexandervolikov/sportiduino/blob/master/Doc/ru/Card.md

mforis, 2018-03-08:
Мг... Станции без проблем берут обычные чипы, но с мастер чипами какие-то проблемы... Самое интересное, что в воскресенье мне нужно будет проводить соревнования с ними... А я тут обновление прошивки затеял... Завтра попробую что-нибудь, а если нет прошью прошивку версии 0.43.1

sembruk, 2018-03-08:
Нужно проверить, что в ячейках для пароля на мастер чипе нули. Для этого нужно прочитать сырые (raw) данные с чипа. Формат ячеек по ссылке ранее.

sembruk, 2018-03-08:
Или ещё есть мысль, что в пароль на станции просочился мусор из EEPROM от прошлой прошивки. Если так, то стирание EEPROM должно помочь. Это можно сделать с помощью Ардуинки в режиме программатора. Про переходник USB-UART не могу сказать.

mforis, 2018-03-08:
Проверил, в там пароли - нули. А как произвести стирание EEPROM, опишите пожалуйста процесс. После стирания придется заново прошивать базы?

sembruk, 2018-03-08:
https://www.arduino.cc/en/Tutorial/EEPROMClear
Вот нашел ввриант. Залить спецпрошивку, которая сотрет eeprom.

sembruk, 2018-03-08:
Железа у меня сейчас нет, проверить не могу.

alexandervolikov, 2018-03-09:
Нашел источник проблемы. Вот с этой прошивкой должно работать. Железа, чтобы проверить, сейчас тоже нет.
[BaseStation.zip](https://github.com/alexandervolikov/sportiduino/files/1795907/BaseStation.zip)

mforis, 2018-03-09:
Отлично, спасибо. Приду домой проверю, после отпишусь)

mforis, 2018-03-09:
Так, вроде бы всё работает, но как сделать, что-бы после отметки ФИНИШ (245), остальные базы уже не реагировали на этот чип, до отчистки?

mforis, 2018-03-09:
Что-то мне эта новая версия софта не очень нравиться... Время за которое участник прошел дистанцию не отображается, не удобно

mforis, 2018-03-09:
Убив два часа, я понял что зря обновил свои станции, правду говорят: - "Старый друг - лучше новых двух.", так и получилось. Я несколько раз до дыр прочитал инструкцию к программе, но так ничего и не понял! Старая версия в сто раз лучше!

closed 2018-03-12 by alexandervolikov