sportiduino/sportiduino #32: Проблема с RFID-модулями

mforis, 2018-04-19:
Заказал 4 RFID-модуля, все из них были заведомо рабочие, один даже впаивал, но после того как отпаял резистор R2 и светодиод, модули перестали работать, в одном пробовал просто отпаять светодиод, после этого модуль тоже прекратил работать... Может вы сталкивались с подобной проблемой?

alexandervolikov, 2018-04-19:
Нет, с такой проблемой не сталкивался. С не работой модуля без светодиода вообще как-то странно. Может статикой пробило от паяльника или ещё как-то (сам я их выламываю кусачками), хотя тоже было бы странно.

mforis, 2018-04-19:
Странно - это не то слово... Я в двух выламывал, потому что забыл отпаять... Блин, жалко считывателей ((

mforis, 2018-11-19:
Вычислил от чего вылетали RFID модули, при пайке паяльником от 220В, его нужно ОБЯЗАТЕЛЬНО ЗАЗЕМЛЯТЬ!!! Или лучше отказаться от сборки станций сетевым паяльником. В идеале паять паяльной станцией или через трафарет. Напишите об этом пожалуйста. И с самой последней прошивкой (MiFare) станция уходит в бутлуп.

alexandervolikov, 2018-11-20:
Не заметил особую чувствительность к статике, паял также 220В паяльником (китайский с терморегулировкой). Также не делал каких-либо усилий, чтобы устранить собственную статику. Скорее всего проблема больше в конкретном паяльнике. Но, конечно, лучше избегать данный фактор, упомяну об этом в мануале.
Позже также посмотрю что может быть не так в прошивке.