sportiduino/sportiduino #33: Нюансы в прошивке 1.3.4

mforis, 2018-06-18:
Доброго времени суток! Обновил я прошивки в своих базах, но нашел некоторые недороботки(по моему мнению):
1. Станция проверки не выполняет свою функцию. Например, я взял чип, чтобы "усыпить" базы, через компьютер его не очищаю, очищаю с помощью базы очистки - всё прошло успешно, прикладываю к базе проверки - успешно, выдаю участнику и опа - чип не отмечается на старте.
2. К примеру, я очистил чип базой очистки, решил проверить работоспособность базовых станций, прикладываю чип к базе очистке - очистка не происходит!
3. Зачем при входе в сон сбрасывать настройки, ведь неудобно, перед каждыми соревнованиями выставлять, что-бы старт был обязательным.

mforis, 2018-06-19:
Хорошо, с этим понятно. Но вот заметил ещё одну проблему, одна из баз, среди ночи начала издавать периодические длинные сигнала, до тех пор, пока не отключил питание.  ли это быть потому, что я временно заменил стабилизатора на AMS1117?

alexandervolikov, 2018-06-19:
Обновил прошивки. в 1.3.5. постарался исправить всё это. Но пока не тестировал.

На счёт издавания длинных сигналов - это станция зависает и перезагружается по кругу. Да, возможно, какие-то проблемы с питанием.

closed 2018-06-19 by alexandervolikov

mforis, 2018-06-19:
Александр, еще такой вопрос, а можно ли подпаять электролит к часам? или только керамика?

alexandervolikov, 2018-06-19:
Думаю, что можно. Удобно smd конденсатор впаивать, там по сути между соседними ножками, вот как я впаивал
[image]

mforis, 2018-06-19:
Протестировал версию 1.3.5. 
1. Настройки не сбивается.
2. Мастер-чипы по прежнему работают некорректно.
3. С базой очистки всё идеально!
4. При прикладывании чипа к базе проверки она издает три сигнала и так на все чипы, в т. ч. мастер-чипы, только- что очищенные и (или) инициализированные.

mforis, 2018-06-20:
Да, нет, обновлял время несколько раз.

alexandervolikov, 2018-06-20:
Да, не учёл один ньюанс для Mifare карт. Сейчас поправил, протестировал, вроде работает корректно.

mforis, 2018-06-20:
Ведь можно же обновить прошивку только в базе проверки, чтобы все не прошивать?

alexandervolikov, 2018-06-20:
Да, конечно. Если использовать станции как базовые, ничего не поменялось с 1.3.0.

mforis, 2018-06-22:
Александр, не могли бы вы скинуть мне файл settings.csv? А то поменял setst на 44, а настройки всё равно сбиваются...

mforis, 2018-06-22:
Хотя... Оказалось, что проблема только в одной базе, наверное не обновил прошивку

mforis, 2018-06-22:
Блин, нет, всё-же скиньте

alexandervolikov, 2018-06-22:
Вот. Только 44 для Mifare тоже что и 12.
[settings.zip](https://github.com/alexandervolikov/sportiduino/files/2129038/settings.zip)

mforis, 2018-06-23:
К сожалению, не помогло. Может проблема в прошивке?

alexandervolikov, 2018-06-23:
Да, проверил, действительно, не работает. Попробую разобраться в чем дело.
А пока просто используйте для базовых станций пошивку 1.3.0, а для станции очистки и проверки 1.3.5.

alexandervolikov, 2018-06-23:
Нашёл источник ошибки. Исправил в 1.3.6. Протестировал, работает, настройки не сбрасываются.
Также для Mifare не работает проверка времени инициализации чипа в режиме базовой станции.