sportiduino/sportiduino #35: Режим ночного ориентирования

mforis, 2018-07-06:
Было бы неплохо, сделать такой режим, в котором бы станция в время работы раз в секунду моргала светодиодом. Участникам соревнований было бы легче найти станцию отметки на контрольном пункте.

alexandervolikov, 2018-07-10:
Батареи будут значительно быстрей садиться. Лучше, по-старинке, использовать хорошие световозвращатели.

mforis, 2018-07-15:
Александр, неделю назад проводил мини рогейн, но всё прошло не очень гладко. Участники прибегали на финиш, я считывал результаты с их чипов, но результаты не совподали с теми, что называли участники. Хотя, говорили, что на всех станциях отметка происходила успешно, то есть был, и звуковой, и световой сигнал. Я решил снять дамп с базовых станции, но в результате, рказалось что данные КП посетили только участники с номерами чипов 1111, 1112, 1113, но фигня в том, что на данных соревнованиях чипы с этими номерами я не использовал.

alexandervolikov, 2018-07-16:
Сложно сказать что произошло не так и откуда появились эти номера. Может с каких-то прошлых манипуляций сохранилось. Переводились ли станции в режим сна перед соревнованиями? И какие были номера присвоены не сработавшим станциями, какой диапазон номеров участников?

mforis, 2018-07-16:
Да, станции переводились в режим сна. Станциям были присвоены номера от 31 до 45 и 100. Номера использованных чипов 21ХХ. Номера участников от 1 до 15.

alexandervolikov, 2018-07-16:
Почему не записались чипы участников понятно: при использовании чипов mifare, номера, чтобы они записывались в память станции, ограничиваются 1500 (иначе они потом не влезут на dump карту). Всё что выше, не записывается. Мой недочёт, нужно это детальнее прописать в документации.
Но откуда взялись 1111 1112 1113 не понятно. Нужно, наверное, потестировать.

mforis, 2018-07-16:
Ага, понятно. Ещё вопрос не по теме, где вы покупали компаунд?

alexandervolikov, 2018-07-16:
вот тут:
https://silagerm.ru/p108938053-silagerm-2104viksint-fasovka.html