sportiduino/sportiduino #37: Настройка станций через NRF24l01

halny123, 2018-07-23:
Александр, как Вы смотрите на возможность добавить в проект настройку станций помимо карточек и через модуль NRF24l01 ?
Это будет намного быстрее и удобнее без увеличения энергопотребления.
А стоимость станции увеличится на 20-30гр.

alexandervolikov, 2018-07-24:
Энергопотребление все же увеличиться. Станции нужно будет периодически включать приёмник, чтобы была возможность передать сигнал.
Думаю, принципиальных изменений в схему вносить уже не буду. Но можете попробовать сделать так самостоятельно.

halny123, 2018-08-03:
Вот первая попытка скрестить с радиомодулем NRF24l01 станции. [Что получилось можно посмотреть тут.](https://youtu.be/FtrvbB7XFUg).
Сейчас устанавливает время, номер станции, задает режим работы стании, задает пароль только не снимает дамп с базовой станции.
Александр если Вас заинтересует и посчитаете нужным внести в проект доработку схема, исходники, плата все есть. Только не пойму как сюда прикреплять файлы, могу только ссылки.

alexandervolikov, 2018-08-03:
Да, довольно удобно получилось.
Внести, конечно, можно. Сделайте pull request в данный репозиторий, добавлю.

(тут одна из инструкций как это сделать: https://git-scm.com/book/ru/v2/GitHub-%D0%92%D0%BD%D0%B5%D1%81%D0%B5%D0%BD%D0%B8%D0%B5-%D1%81%D0%BE%D0%B1%D1%81%D1%82%D0%B2%D0%B5%D0%BD%D0%BD%D0%BE%D0%B3%D0%BE-%D0%B2%D0%BA%D0%BB%D0%B0%D0%B4%D0%B0-%D0%B2-%D0%BF%D1%80%D0%BE%D0%B5%D0%BA%D1%82%D1%8B)

halny123, 2018-10-24:
Вот попробовал сделать как предлагалось отдельным форком.
Здесь описание и видео того, что получилось, и того как я собираю станции.
https://github.com/halny123/sportiduino/blob/master/README.md

closed 2022-01-28 by sembruk