sportiduino/sportiduino #39: Сделать второй светодиод (как в SportIdent)

mforis, 2018-10-08:
Есть предложение добавить в прошивку возможность подключения второго светодиода, который будет загораться с задержкой (к примеру 0,5 секунды) и только в рабочем режиме. Например, при программировании базовых станций он загораться не будет.

alexandervolikov, 2018-10-08:
Добавить можно. Только не понимаю какие это удобства даст. Отличать сигнал при программировании от отметки для того, чтобы убедиться в прохождении процедуры?

mforis, 2018-10-08:
Не совсем, участники соревнований иногда жалуются, что при прохождении отметки не слышно звукового сигнала (в городской черте) и не видно светодиода (за счет того, что он закрывается чипом), ну и в принципе,  то что вы сказали тоже. Просто, по-моему у МК остаются свободные выводы, так что если в это не сложно воплатить, то думаю, это было бы полезно.

alexandervolikov, 2018-10-08:
Понятно.
А если просто сделать мигание тем же светодиодом через данный интервал, но без звукового сигнала?

mforis, 2018-10-08:
Хм... Просто, не будут ли накладываться сигналы друг на друга в случае большого количества спортсменов на КП?

mforis, 2018-10-14:
Александр, обратил внимание, на то, что когда напряжение на батарейках садиться до 3.8 база перестает работать или в лучшем случае, очень плохо считывает чипы.

alexandervolikov, 2018-10-16:
Любопытно. Я вроде тестировал, работало вплоть до 2.8 вольт. Надо будет ещё посмотреть. Что 3.8 вольт весьма странно, там стоит линейный стабилизатор, который всё понижает до 3.3 вольт, хоть 3.8, хоть 5 вольт.

mforis, 2018-10-17:
@alexandervolikov Так, это не первый раз, симптомы такие: станция перестает реагировать на чип, или выходит из спящего режима, но после попытки отметки входит в какой-то цикл (раз в несколько секунд тускло загорается светодиод и иногда тихий звуковой сигнал).

alexandervolikov, 2018-10-18:
дублирующий световой сигнал добавил. Получилось вроде получше. С питанием пока воспроизвести баг не удалось.

mforis, 2018-10-18:
Работают ли базовые станции с этой прошивкой, вместе с базами с прошивкой 1.3?

alexandervolikov, 2018-10-18:
Да, работают. Но полная совместимость утерена в виду изменения байта настроек станции. В частности, изменено назначение битов 4-7. Если раньше настройками станций не пользовались, то разницы никакой. Если ранее использовали настройки, то с новой прошивкой нужно будет обновить их.

mforis, 2018-10-18:
Понял, спасибо! Сегодня попробую записать базовую станцию от блока питания и посмотрю, как она себя будет вести, при разных напряжениях...

alexandervolikov, 2018-10-19:
Потестировал несколько базовых станций, работают вплоть до 2.8 вольт на батарейках. Возможно, в Вашем случае какие-то проблемы со стабилизатором.

mforis, 2018-10-19:
У меня от БП тоже работало вплоть до 2.8, а от батареек, при 3.4-3.8 базовые станции работает не стабильно, то считыватель отрубает, то база в циклическую загрузку уходит...

alexandervolikov, 2018-10-19:
Значит, дело в батарейках. Стоит учесть, что под нагрузкой напряжение на них может довольно значительно упасть. Я тестировал именно с батареек, но они у меня довольно хорошие - Duracel Industrial.

mforis, 2018-10-19:
@alexandervolikov Ну всё, значит нашли в чем проблема... Возможно в будущем, переведу свои станции на АКБ.

mforis, 2018-10-22:
Нашел баг в прошивке 1.4.1: после прохождения отметки станция перезагружается

alexandervolikov, 2018-10-22:
Странно. Но может быть, я её толком не тестировал. Посмотрю в чем может быть дело

mforis, 2018-10-23:
Такой баг только в прошивке 1.4.1, в 1.4.0 станция не перезагружается, но при попытке уменьшить мощность считывателя, станция перестает видеть чипы

alexandervolikov, 2018-10-23:
Спасибо, посмотрю.
Думаю, я сделаю так, чтобы мощность антены можно было настроить только при прошивке.

alexandervolikov, 2018-10-23:
Нашёл возможную проблему перезагрузки. Исправил. 
Также убрал настройки мощности антены внутрь прошивки, так что, если нужна данная настройка, то нужно прописывать перед прошивкой конкретной станции.

mforis, 2018-10-24:
Есть идея, добавить в базовую станцию дисплей, не знаю на сколько это будет полезно. Но на нем могла бы отображаться информации о времени, номере КП и о последнем отметившимся участнике.

mforis, 2018-10-24:
[image]

halny123, 2018-10-24:
Думаю дисплей в базовых станциях излишен, т.к. хрупко и в принципе толку от него не много.
А в мастер станции (если сделать автономное питание) можно выводить время прохождения  дистанции и отметки, для тренировок и быстрой проверки результатов.

В своих станциях добавил геркон, который теперь будит и усыпляет станции. Оказалось очень удобно не нужно ждать 25с чтобы разбудить, и можно усыпить станции в лесу на дистанции при съеме КП без компьютера. Плюс синхронизация времени через радиомодуль порадовала, время на всех станциях синхронизируется точно по нажатию на одну кнопку.