sportiduino/sportiduino #40: Программа для обработки результатов

vasyaabr, 2018-10-24:
Просто как идея: вы можете на заморачиваться на отдельную программу для обработки результатов, всё таки это довольно сложно - предусмотреть много вариантов соревнований, а использовать, например, MeOS (http://www.melin.nu/meos/en/). 

Она очень функциональная и имеет поддержку считывания произвольных систем электронной отметки - надо будет только написать прослойку, которая передаст на нужный сокет пакет отметок с чипа (там есть документация и примеры).

Так же в своё время поступили SFR sytem - тоже написали свой драйвер для MeOS.

alexandervolikov, 2018-10-24:
Сейчас так и сделано с программой SportOrg (http://sportorg.o-ural.ru/). Написан модуль, который там подключили, и теперь можно с ней работать. 
Конечно, я только за, если кто-то напишет драйвер для MeOS, но сам я за это не возьмусь.

alexandervolikov, 2018-10-24:
Да, согласен. К тому же есть станция сопряжения с Bluetooth модулем.

se-ti, 2019-09-25:
> Я думаю, стоит вложить усилия в программу для андроид, так как для компьютера программ вполне достаточно.

@mforis, если тема с андроидом еще интересна,  посмотрите https://sfrreader.github.io/ 
Если заинтересует -- можно обсудить поддержку Sportiduino

mforis, 2019-10-21:
@se-ti , да, стоит с ними обсудить добавление поддержки Sportiduino

se-ti, 2019-10-21:
Не "с ними", а со мной :)

mforis, 2019-10-21:
@se-ti , хорошо, нужно о каких-то условиях договориться?

se-ti, 2019-10-21:
@mforis, ответил в почте

sembruk, 2022-01-28:
В MeOS теперь работает через эмуляцию SPORTident.
[Android приложение](https://github.com/sportiduino/sportiduinoapp) есть.

closed 2022-01-28 by sembruk