sportiduino/sportiduino #48: Non-blocking i2c

fimich1, 2019-01-29:
Библиотека не может быть запущена, потому что пробелы в названии не допускаются
нужно к, примеру так: Non-blocking_i2c

alexandervolikov, 2019-01-29:
Эту библиотеку не совсем обычная. Нужно скопировать папку "Wire" из ней сюда:
C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\

fimich1, 2019-01-29:
Перепрошить уже прошитую станцию удается только после заново залитого бутлоадера. По другому - никак не удалось. Возможно ли в PQ кнопку спец режима для перепрошивки сделать?

alexandervolikov, 2019-01-29:
Прошить станцию можно только в течение 5 секунд после подачи питания. После чего она входит в цикл с режимами сна, в которых прошивка невозможна.
Для этого в ардуино нажимаем прошить, ждём пока пройдёт компиляция (где-то 3/4 полосы загрузки) и до загрузки втыкаем usb-uart и питанием в порт.

mars2860, 2019-01-29:
А я бы отказался от UART-бутлоадера. Не вижу в нём смысла. Вот если бы можно было залить новую прошивку на мастер-чип и обновить станцию - тогда да. А так проще купить себе программатор простой Pickit2 или usbasp и шить новую прошивку без бутлоадера.

fimich1, 2019-01-29:
---Прошить станцию можно только в течение 5 секунд после подачи питания. После чего она ---входит в цикл с режимами сна, в которых прошивка невозможна.
Для этого в ардуино нажимаем прошить, ждём пока пройдёт компиляция (где-то 3/4 полосы загрузки) и до загрузки втыкаем usb-uart и питанием в порт.

вот именно этим способом на своем компе прошить ни разу не удалось, пробовал много

halny123, 2019-01-29:
Изначально у меня тоже была проблема с прошивкой .
Как я ее решил написал тут https://github.com/halny123/sportiduino/tree/master/NRFstation
в разделе О ПРОШИВКЕ.
Теперь шьется без проблем в любой момент.
А чем прошиваете и схема подключения станция - USB-UART какая можно узнать ? И какой преобразователь  USB-UART ?

fimich1, 2019-01-29:
[image]
Все подключено как у Александра, и первый раз все шьется и работает. Не срабатывает именно если нужно перепрошить. 75% компиляции скетча может не смог поймать... Если залить оптибут через ардуино (у меня уно) то все стирается и прошивается. Но если станция залита компаундом, то заново влить бутлоадер еже без расковырки никак. В общем будет проблема в будущем, как придет время менять прошивку. 
[image]

alexandervolikov, 2019-01-29:
Попробуйте прошить, подключив пин DTR вместо RST (они вроде как раз и отличаются наличием дополнительного конденсатора). 
Ну а так да, ещё от схемы в USB-UART зависит гладкость процесса. Я в итоге перешел на схему CP2102, с ним шьётся лучше.

halny123, 2019-01-29:
По первой фото, если у Вас плата Алексея, то насколько я вижу ПЕРЕПУТАНО питание!!!
Не погорел ли после этого стабилизатор 3,3В ?
И остальное подключение не верно.

UART - цвет - Плата Sportiduino
GND -  красный - +5V                    ОШИБКА !!!
+5v -  коричневый - GND              ОШИБКА !!!
RXD    - серый  -RX                         ОШИБКА - нужно взять сигнал с TXD UART
 DTR  - фиолетовый  -TX                 ОШИБКА - нужно взять сигнал с RXD UART
reset - черный - Reset                     ОШИБКА - нужно взять сигнал с DTR  UART

еще у меня китайская плата преобразователя формирует на DTR низкий уровень постоянно при программировании, а должен формироваться импульс сброса. Поэтому проверьте во время программирования какой уровень на ножке Reset . Если там присутствует постоянный 0 то последовательно с линией сброса паяем конденсатор 0,1мк.

И еще на какой скорости происходит программирование ?

fimich1, 2019-01-29:
Фото под таким углом просто, подключено правильно, так как первый раз прошивка проходит: (done)/ станция сигналит. Тут же пытаюсь прошить второй раз - отключаю, нажимаю шить - жду секунд 5 (75 процентов у меня нет, сначала 0 потом сразу 100) - нет
[image]
[image]

. DTR вместо RST пробовал, станция не сигналит второй раз после подачи питания, но все равно не шьется. Конденсатор межу RST и GND тоже не привел к успеху, правда он 10мкф.
[image]

halny123, 2019-01-29:
Давайте попробуем сначала:
Мне кажется мы запутались в терминологии есть два этапа прошивки.
1-й этап -  Загрузка бутлоадера через програматор по SPI интерфейсу - то что у вас на второй картинке, где конденсатор. НО ОН ТАМ НЕНУЖЕН !!
На этом этапе загружается прошивка которая в дальнейшем позволить прошивать МК через UART интерфейс.
Дальше то что делаю я !!!!!
ДЕЙСТВИЯ ОТЛИЧАЮТСЯ ОТ ОПИСАННЫХ АЛЕКСЕЕМ ИЗ ЗА РАЗНЫХ СБОРОК БУТЛОАДЕРОВ !!!

(Можете взять мою прошивку бутлоадера  
https://github.com/halny123/sportiduino/blob/master/NRFstation/Bootloader_Optiboot/bootloaders/myoptiboot/optiboot_atmega328_8M_38K400_D4_3flash.hex
если прошивка пройдет удачно то после подачи питания станция 3раза мигнет светодиодом
 )
Давайте пройдем этот этап.
Прошиваем, подаем питание - и видим мигание 3 раза светодиодом.

Если все будет так переходим ко второму этапу.

2-й этап 
Прошивка через UART 
Для нее нужно правильно настроить среду Ардуино (но это после прохождения 1-ого этапа)

fimich1, 2019-01-29:
Да, картинки не в той последовательности загрузились. Если сначала то так:
1 этап нармально проходит
2 этап нормально проходит.
Станции работают.
В качестве эксперимента предполагаю что в будущем нужно будет поменять прошивку.
Пытаюсь ещё раз прошить. Только 2 этап, так как станция, допустим, залита компаундом.
Вот этот второй этап не проходит.

alexandervolikov, 2019-01-29:
Ещё как вариант. Можно попробовать прошивать станцию во время её работы в "рабочем режиме".  Должно пройти и с залитой компаундом платой. Принцип тот, что в каждом цикле станция засыпает на 250 мс, но время поиска страницы для отметки в начале чипа составляет порядка 160 мс + время сигнала, что чип уже отмечен 1000 мс. Поэтому с вероятностью  80% мы попадаем в режим, когда станция может прошиться.
1. Подсоединить станцию к компьютеру через USB-UART
2. Перевести станцию в рабочий режим - поднести пустой чип.
3. Положить чип под станцию так, чтобы она пробовала сделать туда повторную отметку. Пускай пикает.
4. Пробовать прошить. Если не получилось - повторить.

halny123, 2019-01-29:
Думаю это не выход. Если сигнал сброса правильно формируется то не важно в каком режиме находится станция, после сброса  управление передастся бутлоадеру, а он подождет некоторое время данных от UART.
Изначально нужно смотреть формируется ли сброс правильно.

fimich1, 2019-02-01:
вариант с мигающим чипом не прошел, альтернативный оптибут пробую

halny123, 2019-02-01:
Мой оптибут требует настройки среды Arduino !!!! Я у себя описал какой.
Если нужно подскажу что непонятно.
Но хотелось бы все таки узнать, формируется ли у вас сигнал сброса ?
Что происходит на плате SportIduino на выводе Reset  , до - во время попытки программирования ?
Изначально там должен висеть высокий уровень 3,3В во время начала программирования формируется кратковременно импульс сброса - 0В. Есть ли это ?

fimich1, 2019-02-04:
----На Reset станции подключаем DTR UART. Измерить напряжение на ножке Reset на станции до прошивки и во время прошивки.
Результаты озвучить > до прошивки 3.3.V, во время прошивки 0,   если подключал не DTR a RST то там был логический 1 все время.

---Проверить подключение RX-TX на плате станции и UART. Если правильно подписано все то должно подключаться перекрестно. Но на всякий случай поменять местами и проверить повторно.
> менял конечно

---Ситуация с непрошивкой наблюдается на 1-ой станцией или нескольких ?
> все станции. были сначала на mifare, пришли чипы - перепрошил только опять подключив через spi

fimich1, 2019-02-04:
> На DTR видим формируется сигнал низкого уровня при прошивке, но  постоянный, а должен быть импульс - это хорошо но требует доработки.
> На RST мы не увидели ничего если измеряли цифровым мультиметром - импульс достаточно короткий цифровой прибор может не успевать отработать.  Поэтому пока делать вывод формируется на нем сброс или нет не можем (пока не посмотрим осцилографом).
> Далее делаем небольшую доработку схемы:
> (UART) DTR -----||-----Reset (Sportiduino)
> ^ Конденсатор 0.1мк
> Таким образом формируем короткий импульс сброса МК во время начала программирования.
> Пробуем прошивать.
> Если не получилось повторяем с инверсией RX-TX
> Если не получается то дальше мои предложения будут пробовать мой оптибут под скорость 38к.
> А можно ссылку на файлик который прошиваете через SPI первый раз ?
>
>
> -- С уважением,Ефим Долгов,Оренбург,

осцилографа нет..

sembruk, 2020-01-15:
> Библиотека не может быть запущена, потому что пробелы в названии не допускаются
> нужно к, примеру так: Non-blocking_i2c

Исправлено в v2.6.2

closed 2020-01-15 by sembruk