sportiduino/sportiduino #51: Проблемы с записью чипов базовой станцией

SergeyZaboev, 2019-04-24:
Здравствуйте, собрал мастер-станцию и станции отметки последней версии, прошил их прошивками для чипов ntag. Вывел станцию отметки из сна, приложив мастер-чип для стартовой станции, аналогично сделал для станций отметки. Другому чипу SportIduinoPQ присвоил номер 1. Затем делал отметку на стартовой станции, и проверяю ее наличие в SportIduinoPQ

mars2860, 2019-04-25:
Если вы собрали базовою станцию по последней схеме из репозитория, то это новая схема и для неё ещё нет прошивки под чипы ntag, только mifare

По этой схеме собирали станцию?
https://github.com/alexandervolikov/sportiduino/blob/master/Base%20station/hardware/prod/sportiduino-base-scheme.jpg

SergeyZaboev, 2019-04-25:
Да, по этой схеме. Но, после написания сообщения я перезалил прошивку для mifare и результат почти тот же, мастер-чипы с мастер станции записываются, а вот сами станции при его поднесении отвечают мгновенно длинным сигналом, затем при поднесении чипа участника раздается короткий сигнал, я там понимаю о том что запись отметки произведена, но при считывании в SportIduinoPQ чип остаётся пустым.

mars2860, 2019-04-25:
Скорее всего не настроены часы в базовой станции/номер станции или не проинициализирован чип отметки. В новой версии SportduinoPQ можно создать мастер чип для чтения настроек базовой станции. Создайте этот чип и проверьте настройки базовой станции (часы, номер станции)

SergeyZaboev, 2019-04-25:
Спасибо

SergeyZaboev, 2019-04-25:
Прошил мастер-станцию и станции отметки прошивкой mifare из релиза 1.37, сделал мастер чип для коррекции времени, приложил его, затем для стартовой станции тоже самое. Сделал чип для чтения дампа. Приложил его к станции старта. Считал информацию с чипа в SportIduinoPQ 0.6.0. Получил
Read dump from cp: 13
Total punches: 10
14, 15, 16, 17, 18, 19, 20, 21, 22, 23.
 Ни времени, ни номера стартовой КП.

alexandervolikov, 2019-04-25:
В дампе станции время отметок не хранится, только факт отметки. То есть приведенные данные говорят следующее: на станции под номером 13 отметились чипы с номерами 14, 15, 16, 17, 18, 19, 20, 21, 22, 23.
Время отметки хранится только на чипе.

SergeyZaboev, 2019-04-25:
А я присваивал станции стартовый номер, то есть она должна быть 240. Для эксперимента я работал с одним чипом и стартовой станцией. Все делал после сразу прошивки. То есть не должно было быть на станции отметки этими чипами.

Я конечно извиняюсь за наглость в таких вопросах, но очень хочется воспроизвести работу станций.

SergeyZaboev, 2019-04-25:
Видимо я не разобрался и заказал платы из релиза 1.42. А прошить не знаю точно какой версией прошивки. Вот и задаю вопросы.

mars2860, 2019-04-25:
Покажите фото печатной платы, которую вы собрали и скрин программы SportiduinoPQ

SergeyZaboev, 2019-04-25:
[image]
[image]

mars2860, 2019-04-25:
Вы собрали новую версию печатной платы. С этой платой работает новая версия [прошивки](https://github.com/alexandervolikov/sportiduino/blob/master/Base%20station/firmware/MifareBaseStation/MifareBaseStation.ino.standard.hex). Прошивать её нужно программатором. Никаких бутлоадеров до и после прошивать не надо. Поддерживаются только чипы Mifare. К сожалению, сейчас есть путаница с документацией и тем, что лежит в хеде репозитория. То что в хеде репозитория ещё не вошло в релиз и не документировано. Там есть новые функции и некоторые не соответствия с сигналами, которые подаёт станция по сравнению с текущей документацией. Но задавайте вопросы и я на них отвечу.

Судейскую станцию надо прошить этой [прошивкой](https://github.com/alexandervolikov/sportiduino/blob/master/Master%20station/firmware/usb/MifareMasterStation/MifareMasterStation.ino)

По сборке - светодиод в этой версии не надо так отгинать, устанавливайте его прямо. Черные BLS разъёмы на плату также не надо впаивать, иначе вы не сможете установить батарею.
Программу SportiduinoPQ возьмите отсюда https://github.com/mars2860/SportiduinoPQ. Чтобы прочитать настройки станции нужно создать GetInfo Card. 
[image]

SergeyZaboev, 2019-04-25:
Спасибо, буду пробовать

SergeyZaboev, 2019-04-25:
[image]
Прошил две станции отметки и мастер-станцию, Сделал чип чтения настроек. Получил исходные данные настроек станций. Затем сделал чип установки времени. Приложил его к одной из станций. Станция издала чёткий сигнал (соответственно если задержать чип в поле антенны, то она выдаёт сигнал в соответствии с частотой поиска чипа - я так понимаю). Затем сделал чип чтения настроек снова и приложил его к станции а затем прочитал - ничего не изменилось. На изображении нет, но я тоже самое делал и со второй станцией - результат один.

А может я часы припаял неверно (выемка на микросхеме смотрит на МК)?

SergeyZaboev, 2019-04-25:
Номера станциям также не присваиваются

SergeyZaboev, 2019-04-25:
То есть со стороны SportiduinoPQ и мастер-станции всё работает, а на стороне станции отметки вижу только звуковые и световые сигналы. Даже в чип участника не пишет отметку

mars2860, 2019-04-25:
После сборки у станций отметки пароль (0,0,0). Судя по скрину, вы установили пароль для мастер-чипов (255,255,255), поэтому станция отметки просто отвергает все ваши мастер-чипы. Установите в программе SportiduinoPQ пароль (0,0,0) и нажмите Apply. После этого создавайте мастер-чипы.

[image]

SergeyZaboev, 2019-04-25:
[image]
Ещё тестировал - присвоил станции номер 31, читаю её настройки, в них она уже № 255 и время новое само появляется, на новые настройки станция также не откликается.

SergeyZaboev, 2019-04-25:
Ок, увидел поздно сообщение. Пробую

SergeyZaboev, 2019-04-25:
Заработало, только на одной станции время изменилось на точное, а на второй никак, но уже остальные настройки присваиваются.

Супер!

SergeyZaboev, 2019-04-25:
Пропаял дополнительно ноги у часов и время встало на место.

closed 2019-04-26 by SergeyZaboev

mforis, 2019-04-26:
Добрый день! Какая самая новая прошивка для этих базовых станций?

mforis, 2019-04-26:
[image]

mars2860, 2019-04-26:
Для этой платы новой версии прошивки нет. Последняя для чипов Mifare [здесь](https://github.com/alexandervolikov/sportiduino/tree/master/Base%20station/firmware/MifareBaseStationOld)

mforis, 2019-04-26:
Я и использую Mifare, но платы у меня старые, а хотелось бы использовать новые функции

mars2860, 2019-04-26:
Только если на старой плате резать дорожки и делать проводками перемычки, чтобы на ней работала новая прошивка

mforis, 2019-04-26:
Хорошо, а какие именно отличия в платах?

mars2860, 2019-04-26:
Отличия:
1. Прерывание DS3231 (нога 3) подключено к порту PC3 (нога 26)
2. Порт PD2 (нога 32) подключена к пину RST DS3231 (нога 4)
3. Пин VCC DS3231 (нога 2) отключен от порта PD5 и подключен к цепи +3V3. На порт PD5 теперь подключен пин 32 kHZ DS3231 (нога 1) плюс подтягивающий резистор на 3,3 кОм
4. К порту PD6 (нога 10) подключено прерывание RC522 (6 нога модуля)
5. Вместо биполярного транзистора для управления буззером установлен полевой транзистор
6. Номиналы подтягивающих резисторов I2C уменьшены с 10 кОм до 3,3 кОм

Из всего этого в принципе достаточно на старой плате порезать дорожку от PD5 до DS3231 и кинуть перемыку от 3-го пина DS3231 к порту PC3 (пин 26). Остальные изменения в схеме пока не используются и оставлены для будущих идей. Можно ещё заказать новые печатные платы и перепаять детали. В Китае платы дешевые, мне 50 штук за 800р с копейками сделали