sportiduino/sportiduino #53: Версионирование

askn80, 2019-06-04:
Добрый день!
Прошу помощи:
11 апреля качал здесь gerber-файлы для станций отметки, сейчас по ним изготовлены платы - архив файлов и фото см. по адресу https://cloud.mail.ru/public/3gr6/3nRrjCe8S
В самом архиве gerber-файлы датированы 9 января 2019 года.
Сейчас здесь не могу найти уже ни этих файлов, ни того, что было с ними связано.
А для сборки нужны схема и pcb.

mars2860, 2019-06-04:
Всё на [месте](https://github.com/alexandervolikov/sportiduino/tree/master/Base%20station/hardware/prod) 
Также почитайте [тему](https://github.com/alexandervolikov/sportiduino/issues/51)

askn80, 2019-06-04:
Не похоже что на месте - изготовленные платы по моей ссылке и выложенный там pcb (https://github.com/alexandervolikov/sportiduino/blob/master/Base%20station/hardware/prod/old/PCB.JPG) визуально отличаются

alexandervolikov, 2019-06-04:
Вы смотрите в папке old, нужно смотреть в корневом каталоге:
https://github.com/alexandervolikov/sportiduino/blob/master/Base%20station/hardware/prod/sportiduino-base-assembly.jpg
https://github.com/alexandervolikov/sportiduino/blob/master/Base%20station/hardware/prod/sportiduino-base-scheme.jpg

askn80, 2019-06-05:
Спасибо за информацию, яснее.
Следующий вопрос:
как следует из обсуждения https://github.com/alexandervolikov/sportiduino/issues/51 на сегодняшний день актуальной инструкции по сборке нет, поэтому отталкиваюсь от старой - https://github.com/alexandervolikov/sportiduino/blob/master/Doc/ru/BaseStationAssembly.md
В этой старой инструкции указано, что:
а) спайку изготовленного блока с RFID-модулем следует выполнять ПОСЛЕ прошивки блока
б) а на самом RFID-модуле перед спайкой
> "необходимо выламать/выпаять светодиод и подтягивающий резистор к линии RST, иначе всё это будет сильно потреблять ток во время работы устройства. Также нужно заменить индуктивности L1 и L2 на аналогичные 2.2 мкГн но с большим рабочим током для увеличения мощности антены, например Murata LQH32MN2R2K (Спасибо за совет Александру Якимову)"

Актуальны ли эти рекомендации сейчас - для указанной мной версии станции отметки (под чипы Mifare 1K)?
Допустимо ли будет перепрошивать уже собранные станции (после спайки с RFID-модулем)?

alexandervolikov, 2019-06-05:
Если использовать только чипы Mifare, то выламывать и менять индуктивности не нужно.

Прошивать/перепрошивать можно уже собранные станции (после спайки с RFID модулем). Я давал этот совет с той точки зрения, что, если что-то пойдёт не так, проще отлаживать, перепаивать отдельные компоненты.

askn80, 2019-06-06:
Спасибо, следующий вопрос:
В прежней инструкции по сборке станции отметки https://github.com/alexandervolikov/sportiduino/blob/master/Doc/ru/BaseStationAssembly.md указано:

> Прошивка бутлоадера и загрузка основной прошивки
> Я заливаю буатлодер с помощью другой ардуины, можно делать это с помощью специальных программаторов. Ниже описан процесс с использованием Arduino UNO, точно так же можно использовать и другие Arduino, например, Nano или Pro mini.
> Сначала залить прошивку ArduinoAsISP на Arduino UNO (File - Examlpes - ArduinoAsISP). На Arduino UNO разместить конденсатор 10 мкФ между RST и GND. Подключить к программируемой плате пины 11,12,13 к MOSI, MISO, SCK соответственно. Пин 10 подключить к RST программируемой платы и запитать плату (5 В).
> Установить Optiboot в IDE: вставить URL-адресс: https://github.com/Optiboot/optiboot/releases/download/v6.2/package_optiboot_optiboot-additional_index.json в "Additional Boards Manager URLs" в File-Prefernces После этого прошивки можно установить через меню Tools - Board - Boards Manager Для прошивки нужно выбрать: Board: Optiboot on 32 pins Processor: ATmega328p CPU Speed: 8Mhz (int) Programmer: Arduino As ISP
> Нажать Burn bootloader Если всё нормально, появиться надпись Done burning boatloader Если нет, то где-то допущена ошибка, проверить правильность подключения, пайки и т.п.
> Дальше отключаем UNO от программируемой платы. В IDE меняем программатор на "AVRISP mkII". Загружаем скетч Stantion.ino. Подключить к плате USB-UART переходник. Подключить контакты RST,RX,TX,GND,3.3V, в некоторых переходниках маркировка пинов выполнена нерпавильно и нужно подключть RX-RX, TX-TX, в других RX-TX, TX-RX. Также проблемы могут появиться, если в переходнике не работает пин RST, нужно читать отзывы перед покупкой. Нажать кнопку загрузки скетча.
> Если всё хорошо, то станция должна три раза пикнуть (что говорит о том, что часы работают) и через 5 секунд ещё раз продолжительно пикнуть. Если не так, то попробовать подключить питание 5 вольт напрямую к атмеге (вывод гребенки RFID) и прошить, иногда это помогает. Если нет, проверять пайку.

В обсуждении https://github.com/alexandervolikov/sportiduino/issues/51 указано лаконично:

> С этой платой работает новая версия прошивки - https://github.com/alexandervolikov/sportiduino/blob/master/Base%20station/firmware/MifareBaseStation/MifareBaseStation.ino.standard.hex. Прошивать её нужно программатором. Никаких бутлоадеров до и после прошивать не надо. 

Расшифруйте, пожалуйста, доступнее порядок прошивки (что все-таки изменилось - что из прежней инструкции делать, что не делать).

mars2860, 2019-06-06:
Сделайте как в обсуждении #51 - в станцию отметки залейте программатором прошивку без бутлоадера

mforis, 2019-07-27:
Я так понимаю, что в старые станции новую прошивку нельзя прошить. Так?

mars2860, 2019-07-27:
Можно, только надо в тексте программы заменить #define HW_VERS 2 на #define HW_VERS 1. Скажу, что на старой плате не тестировал. Если вы можете, то испытайте. Также новая прошивка должна одновременно работать как с чипами ntag так и mifare. Если у вас есть чипы ntag, то также просьба потестировать

mforis, 2019-07-27:
Ооо, отлично. Протестирую сегодня или завтра и отпишусь. Прошивать через ARDUINO? И потом будет действовать новая программа SportiduinoPQ?

mforis, 2019-07-27:
Единственное, что у меня чипы mifare, к сожалению, но есть и ntag, могу ими тоже проверить

mars2860, 2019-07-27:
Да, через ардуино

mars2860, 2019-07-27:
И чипами ntag, пожалуйста, проверьте, если есть возможность

mforis, 2019-07-27:
Странно, но в релизе только версия 0.6.0, а 0.7.0 нету в exe. Может сбросите?
[image]

mars2860, 2019-07-27:
опубликовал, можно скачивать

mforis, 2019-07-27:
Спасибо, но теперь в мастер станции звук не идет

mars2860, 2019-07-27:
Наверное, у вас установлен буззер без генератора. В прошивке найдите строчку 

`// the third parameter should be the frequency of your buzzer if you solded the buzzer without a generator else 0
#define beep(ms,n) beep_w(LED,BUZ,0,ms,n)`

И замените её на

`#define beep(ms,n) beep_w(LED,BUZ,3000,ms,n)`

mforis, 2019-07-27:
В общем, не все базы читают чипы после прошивки, наверное, из-за того, что усиление антенны очень большее. И еще базы не видят чипы MIFARE Ultralight, но мастер станция их видит. А вот дату инициализации база очистки не меняет?

mforis, 2019-07-27:
У меня в базах эта прошивка. Может можно как-то в файле прошивки поменять усиление антенны. Сделать его таким же как и в этой.
[MifareBaseStation.zip](https://github.com/alexandervolikov/sportiduino/files/3438527/MifareBaseStation.zip)

mars2860, 2019-07-28:
В базы надо также новую прошивку поставить из репозитория. Усиление антенны тогда через sportiduinopq можно будет менять и чипы ultralight и ntag тогда база будет понимать

mforis, 2019-07-28:
Конечно, я поставил, но говорю, что после установки в некоторых базах чипы вообще не считываются. Можно перед установкой прошивки поменять усиление антенны?

mars2860, 2019-07-29:
Можно. Открываете [Мои документы]\Arduino\sportiduino\sportiduino.h и меняте `#define DEFAULT_ANTENNA_GAIN MAX_ANTENNA_GAIN
` меняете DEFAULT_ANTENNA_GAINT на какой нужно. Также усиление антенны можно через sportiduinopq поменять

mforis, 2019-07-29:
Странно, но не все работают с UART:

> Записываю новый пароль и настройки в базовую станцию по UART
Ошибка: Не совпадает контрольная сумма в ответе <

mforis, 2019-07-29:
И когда прошил она не пикала.

mars2860, 2019-07-29:
Ещё раз прочитайте про прошивку в инструкции по сборке. Там есть обновления. Возможно у вас сейчас не правильно фьюзы выставлены.

mars2860, 2019-07-29:
И если не пикала, то точно также как и для мастер станции надо изменить строчку #define beep(ms,n) beep_w(LED,BUZ,3000,ms,n)

mforis, 2019-07-29:
Она потом пикает, когда отсоединяю UART.

mars2860, 2019-07-29:
Я запутался. Давайте вернемся к усилению антенны и чипам. После изменения усиления антенны, станции по-прежнему не работают с ntag и ultralight?

mforis, 2019-07-29:
Так, смотрите, я прошил две базы со старыми настройками в Arduino IDE (прикладываю скрин)
[image], в одной базе только карточки считывались, больше никакие чипы, в другой ничего не считывалось, в одной базе я установил настройки с помощью UART, а во второй выбивает ошибку, там я чипом изменил усиление, в обоих выставил 23 dB и они считывают всё и вроде нормально работают, а вот третья база прошилась, но не запищала, поэтому я не знаю, что сделать. Эта прошивка по моему мнению лучше чем старая, тут на много громче пищит, вот ну единственное поменялся сигнал повторной отметки.

mars2860, 2019-07-30:
Лучше здесь обсуждать, чтобы потом другие могли посмотреть и разобраться, если у них не будет получаться. Станция, которая не запищала - она только не запищала, а светодиод моргает? И какое место в инструкции вызвало затруднение в понимании?

mforis, 2019-07-31:
Светодиод не моргает. Я, в общем, решил пока те базы, которые прошил на новую, прошить на 1.3.7, но мастер станцию не получается, вылазит эта ошибка:

> Arduino: 1.8.9 (Windows 10), Плата:"Arduino/Genuino Uno"

Изменены опции сборки, пересобираем все
Скетч использует 11028 байт (34%) памяти устройства. Всего доступно 32256 байт.
Глобальные переменные используют 495 байт (24%) динамической памяти, оставляя 1553 байт для локальных переменных. Максимум: 2048 байт.
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x31
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x00
avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0x00
avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0x00
avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0x00
avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0x00
avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0x00
avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0x00
avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0x00
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x00
Проблема загрузки в плату. Помощь по загрузке: http://www.arduino.cc/en/Guide/Troubleshooting#upload .

Этот отчёт будет иметь больше информации с
включенной опцией Файл -> Настройки ->
"Показать подробный вывод во время компиляции"

mars2860, 2019-07-31:
Так мастер-станция это Arduino Nano, а у вас Uno установлено. Возможно ещё надо опцию Old bootloader поставить.

mforis, 2019-07-31:
Да, вы правы, теперь прошилось. Я действительно, невнимательно выставил настройки. А новая прошивка мне очень понравилась, но они плохо почему-то работают в некоторых моих базах.

mforis, 2019-09-08:
Чтобы прошивка новая хорошо работала в старых базах, нужно обновлять загрузчик?

mars2860, 2019-09-09:
У меня нет старой версии платы и не могу сказать точно. При разработке новой версии прошивки старался, чтобы сохранилась совместимость с прошлой версией электрической схемы. Но это всё делалось в теории только по электрической схеме. Тестировать не тестировал со старой платой. По поводу загрузчика, то лучше обновить, на тот, что теперь в репозитории

mforis, 2019-09-09:
Хм... Плохо, потому чтобы обновить загрузчик нужно отпаивать считыватели

mars2860, 2019-09-09:
По идее модуль RC522 никак не должен мешать обновлению бутлоадера. Для надежности, можно на время подтянуть RST-пин RC522 к GND, тогда модуль будет не активен и не будет создавать проблем на шине SPI.

mforis, 2019-09-10:
Arduino: 1.8.9 (Windows 10), Плата:"Arduino/Genuino Uno"

Error reading file (C:\Users\mfori\Desktop\arduino-1.8.9\hardware\arduino\avr\boards.txt:0): Invalid line format, should be 'key=value'

Ошибка компиляции для платы Arduino/Genuino Uno.

Этот отчёт будет иметь больше информации с
включенной опцией Файл -> Настройки ->
"Показать подробный вывод во время компиляции"

mars2860, 2019-09-10:
Расшарьте свой файл C:\Users\mfori\Desktop\arduino-1.8.9\hardware\arduino\avr\boards.txt

mforis, 2019-09-10:
[boards.txt](https://github.com/alexandervolikov/sportiduino/files/3594617/boards.txt)

mars2860, 2019-09-10:
У вас в начале файла какой-то мусор в виде символов с кодом EF,BB,BF. Возьмите мой файл 
[boards.txt](https://github.com/alexandervolikov/sportiduino/files/3594675/boards.txt). Как видите, он отличается от вашего на 4 байта. Если сравнить ваш и мой файл утилитой fc.exe, то вы увидите разницу в начале. И потом при компиляции надо установить плату Sportiduino в настройках, а вы поставили  Arduino/Genuino Uno.

mforis, 2019-09-10:
Спасибо!  Чуть позже попробую залить загрузчик и прошивку. Но существует другая проблема с RC522.  Есть две базы, в которых по какой-то причине перестали работать RC522, если подключить к Arduino Uno, то они подключают, то есть в не пишет, что она не подключена, но на чипы не реагирует. Потом может начат считывать чипы, но при установке на плату Sportiduino, она опять перестает реагировать, уже и Atmega328 менял и загрузчик, но ничего не помогает. Может что-то можно сделать?

mars2860, 2019-09-10:
Тут вроде всё понятно, что-то случилось с модулями RC522. Могу посоветовать прогреть феном микросхему RC522 на модуле, просмотреть под лупой печатные проводники на наличие трещин, просмотреть пайку других деталей на модуле, почистить плату от окислов.

mforis, 2019-09-10:
Arduino: 1.8.9 (Windows 10), Плата:"Sportiduino"

***failed;  
avrdude: verification error, first mismatch at byte 0x0000
         0x00 != 0x3f
avrdude: verification error; content mismatch
Ошибка при записи загрузчика.

Этот отчёт будет иметь больше информации с
включенной опцией Файл -> Настройки ->
"Показать подробный вывод во время компиляции"

mforis, 2019-09-10:
Arduino: 1.8.9 (Windows 10), Плата:"Sportiduino"

avrdude: Expected signature for ATmega328P is 1E 95 0F
         Double check chip, or use -F to override this check.
Ошибка при записи загрузчика.

Этот отчёт будет иметь больше информации с
включенной опцией Файл -> Настройки ->
"Показать подробный вывод во время компиляции"

mforis, 2019-09-10:
Вообще не хочет загружаться

mforis, 2019-09-10:
Arduino: 1.8.9 (Windows 10), Плата:"Sportiduino"

avrdude: verification error, first mismatch at byte 0x7e00
         0x00 != 0x11
avrdude: verification error; content mismatch
Ошибка при записи загрузчика.

Этот отчёт будет иметь больше информации с
включенной опцией Файл -> Настройки ->
"Показать подробный вывод во время компиляции"

mars2860, 2019-09-11:
Либо atmega умерла либо программатор неправильно подключен

mforis, 2019-10-20:
В общем, разобрался в чем была проблема: был плохой контакт. Загрузчик заливается хорошо, без каких-либо проблем. Но вот при прошивке Arduino IDE зависает на загрузке. Все библиотеки установлены, другие прошивки без проблем заливаются, а эта нет... Может что-то нужно сделать?

mforis, 2019-10-21:
Прошил на старый загрузчик, в общем, всё нормально, но нужно как-то полностью отключить усиление антенны, чтобы оно было как в старых прошивках. Можете изменить это в прошивке и скинуть сюда?

mforis, 2019-10-24:
@mars2860,  можете отключить усиление? Или нет возможности этим заниматься?

mars2860, 2019-10-24:
Откройте файл sportiduino.h и в 9 строчке поставьте самостоятельно усиление антенны по умолчанию какое вам нужно.

`#define DEFAULT_ANTENNA_GAIN  MAX_ANTENNA_GAIN`

mforis, 2019-10-24:
К сожалению, это не помогает. Нужно полностью его отключить. В прошивке 1.4.0 уже был такой баг. И в 1.4.1 полностью отключили усиление.

mforis, 2019-10-25:
@halny123, жалениюсожалению, это не помогает (((

Evgen-F, 2019-11-01:
> Arduino: 1.8.9 (Windows 10), Плата:"Arduino/Genuino Uno"
>
> Error reading file (C:\Users\mfori\Desktop\arduino-1.8.9\hardware\arduino\avr\boards.txt:0): Invalid line format, should be 'key=value' [...]

У меня такая же фигня, только замена на выложенный файл boards.txt от mars2860 ничего не дала.

_Arduino: 1.8.10 (Windows 7), Плата:"Sportiduino"

loading hardware from C:\Program Files\Arduino\hardware: loading package arduino: loading platform release: Error reading file (C:\Program Files\Arduino\hardware\arduino\avr\boards.txt:0): Invalid line format, should be 'key=value'

Ошибка компиляции для платы Sportiduino._

Прилагаю файл C:\Program Files\Arduino\hardware\arduino\avr\boards.txt и скрин с ошибкой. Сможете подсказать, что не так?
image.png

mars2860, 2019-11-01:
Выкладывайте свой файл boards.txt прямо из папки C:\Program Files\Arduino\hardware\arduino\avr\

Evgen-F, 2019-11-01:
> Выкладывайте свой файл boards.txt прямо из папки C:\Program Files\Arduino\hardware\arduino\avr\
Так это он и есть. Я не знаю как его выложить непосредственно на Github, без файлхостинга

mars2860, 2019-11-01:
Причина вашей проблемы точная такая же как и у mforis. И файл, то что вы выложили не соответствует тому файлу, что выкладывал я в посте.
> У вас в начале файла какой-то мусор в виде символов с кодом EF,BB,BF. Возьмите мой файл
> [boards.txt](https://github.com/alexandervolikov/sportiduino/files/3594675/boards.txt). Как видите, он [...]

Значит вы как-то неправильно заменили файл boards.txt на тот, что я выкладывал. Есть два варианта решений. Первый - скачайте нормальный текстовый редактор Notepad++. Создайте в Notepad++ новый чистый boards.txt и скопируйте в него содержимое старого boards.txt. Либо скачайте снова тот файл boards.txt, который я выкладывал и замените им свой файл boards.txt

mforis, 2019-11-01:
@mars2860, а по поводу полного отключения усиления сподвижек нет?

Evgen-F, 2019-11-01:
@mars2860, скомпилилось. Походу я по ошибке не ваш файл скачал, а @mforis, поэтому он и не заработал. Не туда тыкнул походу, когда скачивал, сори.

[3 more replies]

closed 2022-01-28 by sembruk