sportiduino/sportiduino #54: Медленная работа станций

SergeyZaboev, 2019-07-19:
Здравствуйте!

Столкнулся с тем, что купив комплект запчастей, собрал вторую партию из 10 базовых станций отметки (Mifare, ver. 105), прошил их прошивкой что и в первой партии (всё работает исправно, организую спортивный лабиринт постоянно), а вновь собранные станции работают, будто наелись каши-тормозянки. Станции читают мастер-чипы очень медленно, но исправно, ставят отметку исправно, но капец как долго. Издают очень протяжённый сигнал диодом и звуковой также.

Проверил все smd-элементы - разложены в соответствии со схемой сборки.

Я не разбираюсь в тонкостях, но у меня на новых станция одно отличие: маркировка пьезоизлучателя HC0903A D8J, а в предыдущих HC0903A D6G. Остальные smd-компоненты из прошлой партии.

mars2860, 2019-07-19:
Сейчас готовится новая версия прошивки, до 31.07.2019 собираюсь выложить апдейт

mars2860, 2019-07-27:
Доступна новая версия прошивки. Также обновилась инструкция по сборке станции. Прочитайте ещё раз про процесс заливки прошивки. Скорее всего у вас сейчас проблема из-за неверно выставленных фьюзов.

closed 2022-01-28 by sembruk