sportiduino/sportiduino #60: Проблема с программированием времени и даты включения на станциях первой версии

Phylosof, 2019-11-05:
Залил новую прошивку на станции версии 1. В целом все работает нормально, кроме программирования времени пробуждения станции. Например: 
- если задать время и дату 17.01.2020 2:40, на самом деле в станцию пишется будильник на 31.12. 2019 2:40
если задать время и дату 30.12.2019 00:00 на самом деле в станцию пишется будильник на 30.11.2019 02:00.

mars2860, 2019-11-06:
Все станции работают по времени UTC. При установке времени, местное время конвертируется во время UTC. Отметка на чип записывается в формате UTC, а при чтении на стороне PC конвертируется в местное время

Phylosof, 2019-11-06:
Но ведь время UTC отличает от местного на 2 часа, Почему тогда разница в 2 недели, а то и в месяц?

mars2860, 2019-11-08:
Не получается у меня воспроизвести подобную ошибку на плате v2, платы v1 нет. На плате v2 всё нормально работает. По теории и на плате v1 тоже должно работать, потому что код записи, хранения и чтения времени будильника одинаковый как для платы v1 так и для платы v2. Как вы смотрите время будильника, записанное в станцию отметки - через SportiduinoPQ и чип чтения состояния?

Phylosof, 2019-11-08:
Да, смотрю через чип состояния. При этом текущее время станции показывает правильно.

mars2860, 2019-11-08:
Возможно проблема из-за библиотеки DS3231. Попробуйте скачать из репозитория библиотеку [DS3231](https://github.com/alexandervolikov/sportiduino/tree/master/Libraries/ds3231-master) и замените этими файлы ваши файлы в папке C:\Users\User\Documents\Arduino\libraries\DS3231. Затем перекомпелируйте проект и обновите прошивку на станции

Phylosof, 2019-11-11:
Проблема не в библиотеке. Я изначально именно эту библиотеку использую.

closed 2019-12-08 by Phylosof