sportiduino/sportiduino #61: Прошивка 2.6.1 - Возросшее потребление в MODE_SLEEP

halny123, 2019-11-09:
Решил перейти с прошивки 104 на последнюю. Заметил неприятную особенность.
В последней прошивке 2.6.1 потребление станций в SLEEP MODE на порядок выше .
Старая прошивка <0.020mA
Новая прошивка ~0.300mA

mforis, 2019-11-09:
[image]
Да, да так и есть. Потребление возрасло...

mars2860, 2019-11-10:
Понятно. По потреблению пока ничего не могу подсказать, потому что у меня нет платы v1. Постараюсь изготовить и решить проблему.

mforis, 2019-11-10:
@mars2860, а это может быть из-за загрузчика? Я бы обновил загрузчик, но в него не хочет заливаться прошивка.

halny123, 2019-11-10:
Думаю дело не в версии платы.  И на платах версии 2 должна наблюдаться такая же картина.
В прошлой версии использовалась другая библиотека для перевода в сон. А в коде перед переходом в сон все ножки мк переключались на выход, отключалось АЦП и схема контроля питания.
//Low all pin for power save
  for (uint8_t i = 0; i <= A5; i++)
  {
    pinMode(i, OUTPUT);
    digitalWrite (i, LOW);  //
  }
ADCSRA = 0;
 MCUSR = 0;
В последней прошивке этого явно нет, и при переходе в режим сна - все это работает и потребляет.

mforis, 2019-11-10:
М-да, даже не знаю, что делать. Обновил во всех базах прошивку и вот не знаю, надолго ли хватит батареек...

halny123, 2019-11-10:
при таком потреблении дней 100

mars2860, 2019-11-10:
В начале кода все неиспользуемые ноги настраиваются как INPUT_PULLUP. АЦП только включается и выключается при проверке уровня заряда батареи и на момент перехода в сон он итак выключен. Плата v2 у меня 90 мкА потребляет. Скорее проблема в настройках портов ввода-вывода. Или плата v1 с этой прошивкой совсем не переходит в режим сна из-за этого условия
`if(rtcAlarmFlag || Serial.available() > 0 || serialWakeupFlag)
    return;
`
Чтобы проверить, переходит плата в режим сна или нет, добавьте в начале кода директиву #define DEBUG.

halny123, 2019-11-10:
Сейчас добавив в наглую " костыль " в
 void sleep(uint16_t ms)
-------
 for (uint8_t i = 0; i <= A5; i++)
  {
    pinMode(i, OUTPUT);
    digitalWrite (i, LOW);  //
  }
 
 pinMode(DS3231_IRQ,INPUT);  // Оставляем вход для срабатывания прерывания от геркона 
// disable ADC
  ADCSRA = 0;
 ---------------
имею 20-30мкА потребления без выключения схема контроля питания.
но вижу что это неправильно т.к. не до конца понимаю работу новой прошивки .

halny123, 2019-11-10:
Плата в режим сна переходит, это хорошо видно прибором по потреблению.

mars2860, 2019-11-10:
> Сейчас добавив в наглую " костыль " в
> void sleep(uint16_t ms)
> +++++
> for (uint8_t i = 0; i <= A5; i++)
> {
> pinMode(i, OUTPUT);
> digitalWrite (i, LOW); //
> }
>
> pinMode(DS3231_IRQ,INPUT); // Оставляем вход [...]

Если помогло, то можно и так оставить. Нельзя только будет прочитать настройки платы по UART, когда она в режиме MODE_SLEEP

mforis, 2019-11-10:
Хорошо, а это не принесет новые баги? И не могли бы сбросить скетч для платы v1 без геркона?

mforis, 2019-11-10:
И вообще стоит ли сейчас прошивать, может лучше дождаться обновления?

halny123, 2019-11-10:
да нет так оставлять нельзя есть баги которые проявляются но объяснить не могу пока. Нужно что то думать. Прошивку пробовать начал только вчера,  Максим а у вас старая прошивка сколько потребляла ?

mforis, 2019-11-10:
В режиме сна 75-120 мкА

halny123, 2019-11-10:
ого (.
Ребята  может создать в телеграмме канал для быстрого общения ?

mforis, 2019-11-10:
Да, кстати, можно. Плюс, я сейчас прошью одну базу на старую прошивку и замерю еще раз, для большей уверенности.

halny123, 2019-11-10:
Вот попробовал в телеграмме канал Sportiduino. Кому интересно попробуйте зайти.

mforis, 2019-11-10:
Я подписался, только пока писать ничего не могу...

halny123, 2019-11-10:
в телеграме выложил видео потребления с двумя прошивками стандартной - 320мкА  , и без АЦП - ~40мкА

mforis, 2019-11-15:
@mars2860 , ну что никаких сподвижек нету? Если нужно, я могу прошивку на своих базах тестировать.

mars2860, 2019-11-15:
Пока этим не занимался. Мне лучше, чтобы плата была под рукой. Постараюсь изготовить ее поскорее

mforis, 2019-11-15:
Ясно, буду ждать...

halny123, 2019-11-16:
Пока такие результаты борьбы с потреблением
Стандартная прошивка - 0,26мА
По аналогии с функцией sleep прошивки 1.6.0
Если перевести перед засыпанием порты на выход и установить 0 
потребление падает до 0,12мА
 
```
 //Low all pin for power save
 for (uint8_t i = 0; i <= A5; i++)
 {
  pinMode(i, OUTPUT);
   digitalWrite (i, LOW);  //
 }
```
если дополнительно выключать АЦП
потребление падает до 0.03mA
```
 // disable ADC
  ADCSRA = 0;
```

отключение системы контроля питания выигрыша не дает (или я не могу замерить <10mkA)

mars2860, 2019-11-16:
Принято. Протестирую ещё с платой версии 2 и добавлю после этого в репозиторий

halny123, 2019-11-17:
Тут на выходных с Максимом потестировали и нашли еще одну неприятность связанную с первой схемой, у меня питание часов немного по другому организованно поэтому проблему не заметил сразу.
Значит по переводу портов в 0. Выяснилось, что все можно и не переводить основное потребление было связано с PD5 через который дополнительно подавалось питание на DS3231. Они и потребляли дополнительно 140мкА. Но после пробуждения 5 порт нужно возвращать в "1", т.к. к нему привязана подтяжка шин SDA SCL.

Сейчас ложим порты с 2 (чтобы оставить возможность конфигурировать станции по UART ) по А5 в ноль и отключаем АЦП - потребление 0,02-0,03мА. После просыпания PD5 возвращаем в 1. Это все для 1 версии платы.

 Где то еще болтаются лишних 10-20мкА (по сравнению со старой прошивкой), но пока больше нет времени искать.  Чтобы отключать brown-out detector нужно немного править используемую библиотеку Adafruit_SleepyDog_Library - но я по крайней мере своим прибором снижения потребления при этом не заметил.
Вечером Максим протестирует на большем количестве плат и режимах и отпишется

halny123, 2019-11-17:
А можно как то лишние посты убрать оставив суть ?

mforis, 2019-11-17:
В общем, протестил, всё работает активный режим, режим ожидания, режим сна. В режиме сна потребление 0.02 мА, но в старой прошивке потребление было 7-10 мкА. Из сна выходят нормально, но как-то долговато, может кажется. Единственное, что смутила одна база потребляет в режиме сна 0.13 мА, возможно проблема в компонентах. Завтра обновлю прошивку во всех базах и тогда уже будет полностью видно, есть ли проблемы.

mars2860, 2019-11-22:
У меня получилось на плате v2 добиться вот такого результата в 6 мкА.
[image]

Но при этом пришлось отпаять ножки VBAT и VCC у DS3231 и сделать VBAT = 3V3, VCC = 0. Часы работают. Если сделать VCC = 3V3, VBAT = 0, то потребление становится 90 мкА. Обидно, кажется в плате v2 я накосячил с расключением питания DS3231, в плате v1 было по-другому. Перелистал весь даташит DS3231 в поисках ответа чем отличатся режим питания по VCC от режима питания по VBAT, что работает дополнительно в VCC и что не работает в VBAT и почему в VCC оно при этом потребляет больше чем в VBAT? Вроде логика, схема термокомпенсации, I2C работают как в VBAT так и в VCC, но почему при этом VCC потребляет больше - не понятно. Ответа в даташите не нашёл

mforis, 2019-11-22:
@mars2860, А без отпайки ножек DS3231 на новой прошивке, вторая версииюя платы сколько потребляет?

halny123, 2019-11-23:
Ребята если хотите переделать плату v2 прошу немного подождать. Сейчас тестирую схему измерения напряжения на батарейках . деталей 2 резистора 1-2 диода и конденсатор, потребление во сне 0 - при 3.3в,  3-5мкА - 5В. Но можно свести и к 0. Если посчитаете актуальным можно будет внести в схему. А то надоело быть в неведении.

mars2860, 2019-11-23:
> 
> 
> @mars2860, А без отпайки ножек DS3231 на новой прошивке, вторая версия платы сколько потребляет? А то я сейчас 10 баз новой версии буду собирать.
88-90 мкА. У меня год станции пролежали и батарейки разрядились в некоторых

sembruk, 2019-12-11:
Протестировал на плате v1. Если переводить ноги перед засыванием в INPUT_PULLUP, потребление в режиме сна 130 мкА (c47822e). Если переводить в OUTPUT-LOW — 11 мкА (9ade4bb).

PR #80 (2020-04-02): New PCB and big firmware update
Fix #61, fix #63, fix #68, fix #69, fix #75, fix #81, fix #84.
closed 2020-04-02 by sembruk