sportiduino/sportiduino #63: Предложение по измерению напряжения на батарее

halny123, 2019-11-24:
Как обещал, результаты экспериментов с измерением напряжения на батарее.
Схема
[image]
Как проводились теоретически расчеты
[vccl.xlsx](https://github.com/alexandervolikov/sportiduino/files/3883342/vccl.xlsx)
Результаты - теория практика при одном диоде (при двух - паразитный ток отсутствует во всем диапазоне возможных входных напряжений от4,8В)
[image]
При питании 5,1В паразитный ток через вход АЦП на практике составил 8мкА (при одном диоде с падением на нем 0,45В)

Скетч
переменные
`//----vcc 
const uint8_t ADC_IN  =  A1;         //  Вход АЦП
const uint8_t ADC_ENABLE  =  A0;     //  Управление вкл. АЦП

const uint32_t R1  =  99000;         //  Верхний резистор (Ом)
const uint32_t R2  =  30200;         //  Нижний резистор (Ом)
const uint8_t  Kol_VDD = 1;          //  Количество диодов
const float KOR_Ubat = -0.17;        //  Корректировка напряжения т.к. падение на диоде не постоянно + ИОН грешен
//----vcc`

Функция

`void  vcc(uint8_t n) {
Serial.print ( "измеряем с нагрузкой - " );  Serial.print ( n );  Serial.println( " раз" ); 
analogReference(INTERNAL);   // Используем внутренний ИОН
pinMode (LED, OUTPUT);
pinMode (BUZ, OUTPUT);

// Включаем делитель
pinMode(ADC_ENABLE,OUTPUT);
digitalWrite (ADC_ENABLE, LOW);

// Переводим на вход для измерения напряжения
pinMode(ADC_IN,INPUT);

 delay (500);

  Serial.print(" без нагрузки 0 - ");
    // преобразуем считанное значение (которое меняется от 0 до 1023) в напряжение (0 – 1.1В):
  float  Ubat_0 = analogRead(ADC_IN)*1.1/1024*(R1+R2)/R2+Kol_VDD*0.6+KOR_Ubat;
  Serial.println(Ubat_0);
 // Нагружаем
 digitalWrite (LED, HIGH);
  delay (1000);
 tone (BUZ, 4000);
 delay(1000);

  for (uint8_t i = 0; i < n; i++) {
    int U_ADC_IN = analogRead(ADC_IN);
    Serial.print(U_ADC_IN);  Serial.print(" - ");
   
   // преобразуем считанное значение (которое меняется от 0 до 1023) внапряжение (0 – 1.1В):
   float Ubat = U_ADC_IN*1.1/1024*(R1+R2)/R2+Kol_VDD*0.6+KOR_Ubat;
   Serial.println(Ubat);
   delay (50);
  }

  // без нагрузки

 noTone(BUZ);
 delay (2000);
  Serial.print(" без нагрузки - ");
    // преобразуем считанное значение (которое меняется от 0 до 1023) в напряжение (0 – 1.1В):
     Ubat_0 = analogRead(ADC_IN)*1.1/1023*(R1+R2)/R2+Kol_VDD*0.6+KOR_Ubat;
  Serial.println(Ubat_0);
 // Отключаем 
 pinMode(ADC_ENABLE,INPUT);  
 digitalWrite (LED, LOW);
} 

//end of vcc`

halny123, 2019-11-24:
Просьба в помощи то что я не в состоянии буду сделать. 
Изменить программу конфигурации SportiduinoPQ
1. Внести возможность при замене батареи сохранять дату замены.
2. Внести возможность вводить и передавать реальное напряжение на батарее для вычисления корректировочного коэффициента. 
3. При считывании мастер чипа состояния - выводить показания батареи без и с нагрузкой.

То что смогу сделать в прошивке и как для себя вижу реализацию

4. Замерять напряжение при каждом задействии бузера - и при просадке ниже 3,8В переводить станцию в режим отметки без звукового сигнала и выставлять флаг о необходимости замены батареи. (Это на любителя - но у себя в прошивке я это точно сделаю т.к. это позволит более-менее гарантированно завершить нормально соревнования если батарея начнет садится в лесу)
5. При выставленном флаге о необходимости замены батареи - при использовании любого мастер чипа - трезвонить "хоть бесконечно" о необходимости ее замены. 

Где то так.
Что думаете ?

mars2860, 2019-11-24:
Мне нравится. Могу расширить SportiduinoPQ под данный функционал. И всё другое, что потребуется

mforis, 2019-11-24:
Было бы очень круто)))

mforis, 2019-11-24:
[image]
Я тут уже маленькую платку сделал, чтобы добавить данную функцию в старые станции.

mars2860, 2019-11-26:
[image]

1. Исправил схему подключения питания для DS3231
2. Добавил схему halny123 для измерения напряжения на батарее
3. Добавил фильтр на ножку RESET, а то при подключении UART переходника станция перезагружается

Просьба проверить и написать замечания

mforis, 2019-11-26:
@mars2860 , а что там с программой Sportiduino PQ?

halny123, 2019-11-26:
R1\R3 наверное можно посадить на +3.3V, а Vcc часов на землю. Все равно он большую часть времени там и находится для уменьшения потребления. + Не нужно будет дергать лишний раз при просыпании ногу.
И предложение - когда будете новую печатку делать предусмотреть отводы хотя бы от некоторых неиспользуемых пинов МК и закончить их переходным отверстием. Тогда в случае доработок легче будет подключаться.

halny123, 2019-11-26:
Антон проверь если есть возможность падение на диодах при малых токах порядка 10мкА. Как в делителе. Последний раз я пробовал делитель 270к\68к падение на диодах было порядка 0,4В , ставил что под руку попалось - 2n4148. Интересно на сколько стабильно будет это падение при малых токах. Может вообще их выбросить пожертвовав ~2мкА потребления. 
Критично при измерении было выдержать время после переключения в режим измерения не менее 25мс, иначе первое измерение идет с большой ошибкой, пока порты переключались.

mars2860, 2019-11-26:
Думаю диоды не стоит выбрасывать и на печатной плате сделать под них посадочные места. Тогда если с диодами будет плохо, то вместо них можно будет просто впаивать перемычки. Также по питания DS3231 добавил перемычек для выбора питания шины I2C от 3.3В или порта МК, также перемычка для выбора питания DS3231 от порта МК или без питания VCC. Также решил объединить разъём для программатора и разъём для UART в один разъём. А то когда 2 разъёма, то можно запутаться куда подсоединять UART переходник, а куда программатор. С одним разъёмом проще будет понять, просто разные жгуты будут для программатора и переходника UART. В итоге получается такая схема. Печатка по габаритам будет как версия 1. Если есть ещё предложения и замечания, то говорите.

[image]

halny123, 2019-11-27:
Вроде все так.

mforis, 2019-12-01:
@mars2860, Антон, может всё-таки подскажите когда планируется обновление Sportiduino PQ, а то хочется узнать прошивать базы на новую прошивку или нет

mars2860, 2019-12-02:
> 
> 
> @mars2860, Антон, может всё-таки подскажите когда планируется обновление Sportiduino PQ, а то хочется узнать прошивать базы на новую прошивку или нет

Наверное, ближе к НГ или в период праздников, когда будет свободное время.

mforis, 2019-12-05:
@mars2860, а можете скинуть схему в электронном виде? Желательно целым проектом...

mars2860, 2019-12-05:
> 
> 
> @mars2860, а можете скинуть схему в электронном виде? Желательно целым проектом...

Залил в репозиторий. Смотрите ветку develop. Печатную плату тоже уже поправил.

mforis, 2019-12-05:
@mars2860, а что поменяли в прошивке? И будет ли работать новая прошивка на плате версии 3?

mars2860, 2019-12-05:
Пока не будет. Работа ещё в процессе. В 3 версии пришлось поменять некоторые пины для удобства разводки ПП. В прошивку данные изменения ещё не внес.

mforis, 2019-12-05:
@mars2860, Хорошо, а какие изменения? Стоит ли обновлять прошивку перед проведением соревнований?

mars2860, 2019-12-05:
Изменения касаются энергосбережения в режиме сна, то что здесь обсуждали. Обновлять не стоит, так как плохо протестировано.

halny123, 2019-12-06:
Антон, по поводу печатки новой. Вы сделали размер полностью под корпус и залили все сплошной землей. Это очень негативно скажется на дальности обнаружения меток вплоть до 0, т.к. будет экранировать поле катушки.
В самой первой версии я на такие грабли наткнулся.

halny123, 2019-12-06:
Я заказал уже платки, размер полностью под корпус (мне так удобно), но можно и уменьшить в два раза, если после тестирования все будет нормально выложу у себя может кому пригодится. 
[image]

mars2860, 2019-12-06:
> 
> 
> Антон, по поводу печатки новой. Вы сделали размер полностью под корпус и залили все сплошной землей. Это очень негативно скажется на дальности обнаружения меток вплоть до 0, т.к. будет экранировать поле катушки.
> В самой первой версии я на такие грабли наткнулся.

Спасибо! Убрал области металлизации

mars2860, 2019-12-06:
> 
> 
> Я заказал уже платки, размер полностью под корпус (мне так удобно), но можно и уменьшить в два раза, если после тестирования все будет нормально выложу у себя может кому пригодится.
> [image]

Хорошее расположение компонентов. Действительно надо идти в ширину, а не длину, чтобы под антенной не было компонентов и проводников. Единственное, только номер версии указать не 2.6.1, а как-нибудь sportiduino hardware v3a, чтобы не возникла путаница. Первая цифра порядковый номер желаза, вторая порядковый номер прошивки. И как будет подключаться программатор?

halny123, 2019-12-06:
Программатором после запайки всего кроме RFID,  сажусь на контакты RFID - это однократная операция, далее все шью через UART.
Плату для себя делал, версию ставил равную версии прошивке на момент разработки. Для меня не принципиально. Есть партия первой версии - это будет вторая. Для нашего клуба 50 станций будет более чем достаточно - запутаться будет тяжело.
В этой версии пришлось на низ вынести немного компонентов 
https://easyeda.com/burenko/sportiduino_v2-6-1 - вот то что сейчас получается.

Антон, вопрос хоть и не по теме - в новой прошивке переработали процедуру очистки чипа ?
А то на старой прошивке очистка идет 5-7с (NTAG215) , а на новой 1-2с. Заметив это сильно удивился но не посмотрел что происходит при очистке в новой прошивке.

mars2860, 2019-12-06:
По сравнению с версией 1.4 - да. По сравнению с версией 2.6.0 - нет.

PR #80 (2020-04-02): New PCB and big firmware update
Fix #61, fix #63, fix #68, fix #69, fix #75, fix #81, fix #84.
closed 2020-04-02 by sembruk