sportiduino/sportiduino #66: Optiboot не хочет заливать прошивку

sembruk, 2019-12-08:
Проблема на этапе заливки прошивки через USB-UART. Проявляется всех трёх платак, которые спаял.
Первый байт от avrdude проходит. Второй только иногда. 
Пробовал разные Optiboot-ы: и hex из этого репозитория, самосборный из мастера, и v8.0
USB-UART на чипе CP2102.
Плата v1 собственной разводки. Наверное, дело в ней.
Может кто сталкивался с похожей проблемой?

sembruk, 2019-12-08:
```
/opt/arduino-1.8.10/hardware/tools/avr/bin/avrdude -C/opt/arduino-1.8.10/hardware/tools/avr/etc/avrdude.conf -patmega328p -carduino -P/dev/ttyUSB0 -b115200 -D -Uflash:w:/tmp/arduino_build_613570/BaseStation.ino.hex:i -v -v -v -v
<…>
avrdude: Send: 0 [30]   [20] 
avrdude: Send: 0 [30]   [20] 
avrdude: Send: 0 [30]   [20] 
avrdude: ser_recv(): programmer is not responding
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x00
avrdude: Send: 0 [30]   [20] 
avrdude: ser_recv(): programmer is not responding
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x00
avrdude: Send: 0 [30]   [20] 
avrdude: ser_recv(): programmer is not responding
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0x00
avrdude: Send: 0 [30]   [20] 
avrdude: Recv: . [14] 
avrdude: Recv: . [10] 
<…>
avrdude: Send: A [41] . [80]   [20] 
avrdude: ser_recv(): programmer is not responding
avrdude: stk500_recv(): programmer is not responding
avrdude: Send: A [41] . [81]   [20] 
avrdude: ser_recv(): programmer is not responding
avrdude: stk500_recv(): programmer is not responding
avrdude: Send: A [41] . [82]   [20] 
avrdude: ser_recv(): programmer is not responding
avrdude: stk500_recv(): programmer is not responding
avrdude: Send: A [41] . [98]   [20] 
avrdude: ser_recv(): programmer is not responding
avrdude: stk500_recv(): programmer is not responding
<…>
avrdude: Send: A [41] . [84]   [20] 
avrdude: ser_recv(): programmer is not responding
avrdude: stk500_recv(): programmer is not responding
avrdude: Send: A [41] . [85]   [20] 
avrdude: ser_recv(): programmer is not responding
avrdude: stk500_recv(): programmer is not responding
avrdude: Send: A [41] . [86]   [20] 
avrdude: ser_recv(): programmer is not responding
avrdude: stk500_recv(): programmer is not responding
avrdude: Send: A [41] . [87]   [20] 
avrdude: ser_recv(): programmer is not responding
avrdude: stk500_recv(): programmer is not responding
avrdude: Send: A [41] . [89]   [20] 
avrdude: ser_recv(): programmer is not responding
avrdude: stk500_recv(): programmer is not responding
<…>
avrdude: Send: A [41] . [81]   [20] 
avrdude: ser_recv(): programmer is not responding
avrdude: stk500_recv(): programmer is not responding
avrdude: Send: A [41] . [82]   [20] 
avrdude: ser_recv(): programmer is not responding
avrdude: stk500_recv(): programmer is not responding
avrdude: Send: B [42] . [86] . [00] . [00] . [01] . [01] . [01] . [01] . [03] . [ff] . [ff] . [ff] . [ff] . [00] . [80] . [04] . [00] . [00] . [00] . [80] . [00]   [20] 
avrdude: ser_recv(): programmer is not responding
avrdude: stk500_recv(): programmer is not responding
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.

avrdude: Send: Q [51]   [20] 
avrdude: Recv: . [14] 
avrdude: Recv: . [10] 
```

mars2860, 2019-12-08:
А конденсатор последовательный между DTR сигналом USB-UART и ножкой RST микроконтроллера есть?

mforis, 2019-12-08:
@mars2860 , у меня так-же после обновления на загрузчик, который в репозитории. Хотя конденсатор есть... И на старом загрузчике без проблем шьётся. Кстати, чем отличается загрузчик от обычного optiboot?

halny123, 2019-12-08:
Если не шьется никак то смотреть формирование сигнала сброса как написал Антон.
Если нестабильно шьется - я описал у себя как поборол. https://github.com/halny123/sportiduino/tree/master/NRFstation - в самом низу пункт ПО ПРОШИВКЕ OPTIBOOT :

halny123, 2019-12-08:
Если будут вопросы - в телеграмме есть группа https://t.me/Sportiduino , где смогу более менее быстро ответить.

sembruk, 2019-12-08:
О! Спасибо за ответы! Оперативно.
@mars2860 , конденсатор есть, но пользоваться выходом DTR перестал пока. Так как на МК только загрузчик. Собрал Optiboot c индикацией на светодиод (D4). Вижу что, загрузчик перезагружает МК каждую секунду, как и надо.
Залил на плату скетч SerialEvent через программатор. Да, проблема в получении последовательности байтов на МК, если они идут по шине подряд.
Но осциллограф говорит, что всё хорошо.
Может не хватает бита чётности?

sembruk, 2019-12-08:
> Если не шьется никак то смотреть формирование сигнала сброса как написал Антон.
> Если нестабильно шьется - я описал у себя как поборол. https://github.com/halny123/sportiduino/tree/master/NRFstation - в самом низу пункт ПО ПРОШИВКЕ OPTIBOOT :

@halny123 , прошивка заливается! Спасибо огромное! Надо будет эти загрузчики добавить в этот репозиторий.

sembruk, 2019-12-08:
Закрою issue завтра pull request-ом.

PR #67 (2019-12-09): Updated Optiboot bootloader hex files
Now it works on 38400 baudrate.
This fix #66 issue
closed 2019-12-09 by mars2860