sportiduino/sportiduino #68: Разъём программирования с защитой от неправильного включения

sembruk, 2019-12-10:
Предлагаю в новой версии платы использовать разъём типа [IDC-10MS](https://www.chipdip.ru/product/idc-10ms) (вилка на плату). Ответная часть на шлейф — [IDC-10F](https://www.chipdip.ru/product/idc-10f) монтируется без пайки.
Он имеет шаг 2,54 мм. Поэтому, если нет таких под рукой, на плату можно запаивать разъёмы старого типа.

mars2860, 2019-12-10:
Да, идея правильная. Сам об этом думал, чтобы разъём под программирование был с ключом. Хотел найти, что-нибудь поменьше IDC-10 и такое же доступное, но не нашёл. Если ставить IDC-10, то получается всё очень плотно. Расстояние между стойками корпуса по ширине 33.5мм минус 12мм бузер, остаётся 21.5 мм. Сам IDC-10 = 20.5мм. В принципе влезет, если ставить всё с одной стороны.

sembruk, 2019-12-10:
А если развернуть разъём вдоль и расположить над RFID модулем? Ножки разъёма не должны в него упираться.

sembruk, 2019-12-10:
Также предлагаю расположить RX и TX на разъёме рядом.

halny123, 2019-12-10:
Если хотите разъем, то только с верху или низу платы, иначе будут проблемы поместить бокс с батарейками. Он просто не влезет.
Наша первая версия плат разъем слева (правда не выводил под программатор пины мне показалось они лишние). И предложение все таки избавиться от светодиода выводного. Очень не технологично и не ремонтопригодно.

[image]

PR #80 (2020-04-02): New PCB and big firmware update
Fix #61, fix #63, fix #68, fix #69, fix #75, fix #81, fix #84.
closed 2020-04-02 by sembruk