sportiduino/sportiduino #69: Версия платы с внешней памятью

sembruk, 2019-12-10:
Планирую в ближайшее время развести плату v3 с посадочным местом под чип 24LC256.
Ветка ext-eeprom.

mars2860, 2019-12-10:
Насколько понимаю внешняя EEPROM будет использоваться под журнал отметок или есть ещё какие идеи?

sembruk, 2019-12-10:
Да, для этого.

sembruk, 2019-12-10:
Добавил память в схему и переделал подключение разъёма программирования
[image]

mars2860, 2019-12-11:
Вроде всё норм. Версию можно просто 3 без m, всё равно прошлая схема ещё не вошла в релиз. Тогда эту схему добавим в новый релиз и она будет 3-й версией. И может ещё стоит подумать и как-то привести к общему знаменателю идею Виталия (@halny123) с радиомодулем и текущую схему и создать один официальный релиз. Возможно, вместо NRF радиомодуля использовать bluetooth модуль HC-08. Золотой Маршрут выбрал в качестве отметки систему "Фрактал", там вроде bluetooth. На будущее возможно можно будет сделать совместимость с этой системой. Кто что думает?

sembruk, 2019-12-11:
В Телеграм-чате Максим предложил хорошую идею: пробуждать станцию по геркону. Можно пробуждать по геркону и чипу.
При этом не нужно каждые 25 секунд включать RFID модуль и в режиме сна можно существенно понизить потребление.

sembruk, 2019-12-11:
Я считаю, что совместимость с Фракталом не нужна. По крайней мере пока.

mars2860, 2019-12-11:
Можно сделать так. В схему добавить геркон как у Виталия. На разъёме программирования X2 выкинуть VBAT и вместо него подключить какой-либо порт МК, например, PB0 или PD7 - это будет Chip Select для другого устройства SPI. По желанию, пользователь пусть хочет ставит NRF или Bluetooth или вообще ничего. Радиомодуль будет ставиться дополнительно с торца корпуса и подключаться к разъёму программирования, ведь этот разъём всё равно пустует, когда корпус закрыт. Надо тебе обновить прошивку - открутил крышку, вытащил радиомодуль из разъёма программирования, перепрошил, поставил радиомодуль обратно. Или даже для пользователей с радиомодулем бутлоадер сделать для бесконтактного обновления прошивки. Идей много. И сейчас для этого в принципе надо только геркон добавить и вместо VBAT какой-нибудь порт.

mforis, 2019-12-11:
Семён прав, ни вижу особого смысла использовать радиомодули, так как возастет потребление. Да и вообще не вижу смысла склеивать две системы. А вот геркон вещь хорошая. А по поводу NRF его можно использовать для реализации бесконтактной отметки, то есть сделать активный чип, как вариант, хотя это достаточно дорого + увеличивает потребление

halny123, 2019-12-11:
Ребята может в чат перелезем по обсуждению ? Я могу рассказать что есть у меня сейчас чем доволен чем нет . А сюда уже выложить выжимки из обсуждений. А то в проблемах будет много мусора.
Кратко:
- Герконом доволен полностью не жалею что сделал. Просыпаюсь мгновенно. Если нужно можно сделать засыпание.
- Радиомодуль - как таковой оказался не востребован, побаловался и все. для настройки все же быстрее чипами оказалось. Была задумка использовать для перепрошивки, optiboot позволяет OTA прошивку делать. Но это настолько редко, что легче открутить крышку и нормально перешить. Поэтому отказался от него.
- Внешняя память - не увидел необходимости вообще. Критично только наличие номера чипа - время не критично. Если уж нужно примерно знать время отметки - можно вычислить по соседним чипам. (Отказался)
- Сейчас задумка отказаться от чипов для программирования за счет использования на мастер станции модуля PN532 - который должен поддерживать режим эмуляции чипа. Но пока детали в пути сказать работает или нет задумка не могу. Тогда необходимость в радиомодуле отпадет вообще.
Это все только мое мнение у других требования к отметке могут быть иные.

halny123, 2019-12-11:
> Можно сделать так. В схему добавить геркон как у Виталия. На разъёме программирования X2 выкинуть VBAT 
Я бы предложил выбросить Vcc а питание заводить все таки на VBAT.
Не дай бог кто то воткнет что то больше 3.3В. Пусть лучше сгорит стабилизатор чем что то еще.

sembruk, 2019-12-11:
Ещё хочу убрать фильтр с линии Reset Атмеги. Так как сигнал с DTR идёт через конденсатор, то имульс получается очень коротким. Будет ли с фильтром работать обновление прошивки?

mars2860, 2019-12-11:
> 
> 
> Ещё хочу убрать фильтр с линии Reset Атмеги. Так как сигнал с DTR идёт через конденсатор, то имульс получается очень коротким. Будет ли с фильтром работать обновление прошивки?

Ещё не проверял как эта цепь будет работать, но всё же выбрасывать думаю не стоит. Если это будет мешать программированию, то можно просто не распаивать этот фильтр.

sembruk, 2019-12-12:
[image]

mars2860, 2019-12-12:
PC3 зарезервирована под управление питанием DS3231. Здесь я уже сделал ошибку, когда VCC была жестко к 3v3. Сейчас мы хотим попробовать VCC=0, R1,R3 = 3V3, но неизвестно пока какое будет потребление, поэтому мной были добавлены печатные джамперы, которые позволяют вернуться к проверенному решению из версии 1. Лучше PB6 или PB7 задействовать. Кварцы ставить точно не будем.

sembruk, 2020-01-16:
> Ещё не проверял как эта цепь будет работать, но всё же выбрасывать думаю не стоит. Если это будет мешать программированию, то можно просто не распаивать этот фильтр.

Да, с LC фильтром на RST не обновляется прошивка. Думаю, всё-таки его убрать с платы. А для предотвращения перезагрузки станции при поключении USB-UART можно физически отключать DTR.

PR #80 (2020-04-02): New PCB and big firmware update
Fix #61, fix #63, fix #68, fix #69, fix #75, fix #81, fix #84.
closed 2020-04-02 by sembruk