sportiduino/sportiduino #75: Пробуждать станцию с помощью магнита

sembruk, 2020-01-15:
или магнита и чипа.
На плате v3 для этого предусмотрен геркон.

halny123, 2020-01-16:
Осмелюсь предложить следующий алгоритм, который позволит и пробуждать и усыплять станцию магнитом.
При однократном касании магнитом - станция пробуждается (через перегрузку). 
Если магнит удерживать - после перегрузки в setup - проверять состояние геркона - если он будет оставаться замкнут - усыплять станцию.

sembruk, 2020-01-16:
Усыпление станции магнитом опасно: могут намеренно или случайно усыпыть её в лесу.

sembruk, 2020-03-19:
0361a05
Теперь по умолчанию станция опрашивает чип каждые 25 сек (как и раньше). Если сработал геркон, чип опрашивается немедленно.
Если определить NO_POLL_CARDS_IN_SLEEP_MODE в прошивке, станция не будет опрашивать чип, пока не сработает геркон.

PR #80 (2020-04-02): New PCB and big firmware update
Fix #61, fix #63, fix #68, fix #69, fix #75, fix #81, fix #84.
closed 2020-04-02 by sembruk