sportiduino/sportiduino #76: Мастер-чип для одновременной записи номера станции и времени

Pavel-Y, 2020-01-18:
User story:
перед соревнованиями я 
1) раскладываю станции на столе
2) на каждой устанавливаю время
3) на каждой устанавливаю номер
4) обычным чипом прохожу по всем станциям и считываю результат
5) усыпляю каждую станцию
6) убираю станции

Предлагаю совместить шаги 2 и 3.

halny123, 2020-01-19:
А зачем каждый раз устанавливать номер станции ?
Время понятно, из за плохих часов уходит. Но номер вроде меняется крайне редко.

Pavel-Y, 2020-01-20:
Зачем: 
1) Я беру станции у приятеля, а не из шкафа. Мне проще прикрепить наклейку с нужным номером, чем искать нужную станцию, которой может и не быть.
2) В рогейне есть КП с номерами типа 67, 82, 91. Таких станций в наборе скорее всего нет. К тому же в одном рогейне три номера 3х, а в другом четыре.
3) Если я взял станцию у приятеля и на ней наклейка 42, нет гарантий, что в ее памяти тоже 42. Надо проверять или сразу перезаписывать.

closed 2022-01-28 by sembruk