sportiduino/sportiduino #77: Станции пишут на чип время на 3 часа позже чем при их инициализации

Pavel-Y, 2020-01-21:
Воспроизводится так: 
- создаем мастер чип с неким временем HH:MM:SS по протоколу https://github.com/sportiduino/sportiduino/blob/master/Doc/ru/Card.md
- прикладываем к станции
- создаем чип участника 
- прикладываем к станции
- станция пишет на чип unixtime, который равен (HH+3):MM:xx  (если всё уложилось в 1 минуту)
- на чипе unixtime инициализации чипа и unixtime отметки станции тоже отличаются примерно на 3 часа, хотя между ними прошло несколько секунд (станции лежат на столе, а не в поле).

Найти эту особенность помог просмотр raw data на чипах при разработке софта для смартфона.

Чем это плохо: 
- кажется, в код станции где-то зашита таймзона UTC+3. Поэтому если в другой TZ писать в протокол соревнований время взятия КП, оно будет неправильным.
- разный формат времени в разных местах (unixtime и ММ ГГ ДД ЧЧ ММ СС), не понятно, timezone aware или нет. 
- кто-то переводил документацию на английский. С такой таймзоной нас не поймут :)
- при масс-старте (без отметки на стартовой станции) надо будет этот факт учитывать.

На что не влияет: 
- если публиковать только разницу с временем отметки на стартовой станции, это не видно. 

Предложение: 
- работать со временем в станциях и чипах в UTC. Дальше конечный пользователь без проблем конвертирует его в свою TZ.

sembruk, 2020-01-21:
@Pavel-Y, какие версии прошивки и SportiduinoPQ использовались?

Pavel-Y, 2020-01-21:
2.6.2, но кажется это моя ошибка. Закрываю

closed 2020-01-21 by Pavel-Y