sportiduino/sportiduino #79: Станции КП пишут отметки на чип без номера. Не должны?

Pavel-Y, 2020-01-22:
Станции КП пишут отметки на чип с нулевыми 4й и 5й страницами (скорее всего это использованный мастер чип). 

Прошивка (4 стр. чипа, последний байт) 0x67.

Имею мнение, что такой чип не должен уйти на дистанцию, чтобы не возиться с ним на финише, а должен быть отловлен на стартовой станции (нет сигнала о записи) и заменен или возвращен на станцию записи номера.

Или большей проблемой станет участник, прошедший всю дистанцию с молчавшими станциями? 

А что происходит, если страница перестает читаться? С нее приходят нули или ошибка ввода-вывода?

sembruk, 2020-01-22:
Прошивка 1.3.x (0x67) старая. В версиях ≥ 1.6.0 станция не должна записывать отметку на такой чип.

Pavel-Y, 2020-01-22:
Для перепрошивки станции необходимо вскрыть коробочку и подключиться к торчащему из компаунда разъему, так?

sembruk, 2020-01-22:
Проводились ли старты не знаю. Ошибки в прошивке есть. Планирую активно в этом году использовать станции >= 3.7.
У Sportiduino есть сообщество пользователей. Общение проходит в чате Телеграм [t.me/sportiduino](t.me/sportiduino). Там быстрее отвечают на вопросы.

closed 2020-02-03 by sembruk