sportiduino/sportiduino #83: Config by UART

arashipl, 2020-02-04:
I am trying to put together a set of control station on my own PCB's customized to fit cases I used for a different project and use a 16350 Li-Ion cells I have instead of 3x R6 as an option.
I mixed up a different versions of schematics, so my design is slightly different and I faced quite a few complications, but it seems all is on a good way, thanks to Google Translate I found halny123's test app, which helped me diagnose my stations. So I've arrived to the point where I want to configure and prepare stations to be used and I tried to config stations via uart as a part of programming procedure (flash bootloader, use UART to test assembly, then upload basestation sketch and then configure station withuout the hassle of using programming via rfid chips.
But no matter what I try I always get "Error: No response from the base station" info in SportiduinoPQ 0.7.1.
Is this a bug or I missed something? I use the same cable for SportiduinoPQ that I use for programming with ArduinoIDE and for TestStationOK debug communication.  - and there are no problems whatsoever.

mars2860, 2020-02-04:
Don't use a cable for programming with ArduinoIDE because it has DTR signal which resets MPU before programming. You need other cable which has just 3 wire (rx,tx,gnd)

arashipl, 2020-02-04:
Thank you for the answer, I just assumed that DTR is left idle when port is configured with no flow control. 
Just for the record - I admire amount of work that went into this system and I offer whatever help I may need. SportIduino is getting a foothold in Poland :)