sportiduino/sportiduino #86: Проверка записи данных на чипы в 3ей версии прошивки на базовую станцию

Phylosof, 2020-03-08:
Порылся недавече в библиотеке rfid (для 3ей версии прошивки) и сравнил функцию ntagWrite из первой версии BaseStation.ino с функцией ntagCardPageWrite в rfid.cpp и заметил существенное отличие. В ntagCardPageWrite нет проверки записи на чип путем считывания и сравнения данных из буфера, что существенно снижает надежность отметки. Может стоило бы сохранить сей полезной момент в новой версии?

halny123, 2020-03-09:
Дим ты просил показать где происходит проверка записи отметки, последние 10 строк
функции

bool ntagWrite (uint8_t *dataBlock, uint8_t pageAdr){

const uint8_t sizePageNtag = 4;
status = (MFRC522::StatusCode) mfrc522.MIFARE_Ultralight_Write(pageAdr,
dataBlock, sizePageNtag);
if (status != MFRC522::STATUS_OK) {
return false;
}

uint8_t buffer[18];
uint8_t size = sizeof(buffer);

status = (MFRC522::StatusCode) mfrc522.MIFARE_Read(pageAdr, buffer, &size);
if (status != MFRC522::STATUS_OK) {
return false;
}

for (uint8_t i = 0; i < 4; i++) {
if (dataBlock[i]!=buffer[i]) return false;
}

return true;

вот функция записи на чип все операции в конце верефицируются чтением - это наша
теперешняя прошивка

  08.03.2020, 19:58, Phylosof <notifications@github.com>

  Порылся недавече в библиотеке rfid (для 3ей версии прошивки) и сравнил
  функцию ntagWrite из первой версии BaseStation.ino с функцией
  ntagCardPageWrite в rfid.cpp и заметил существенное отличие. В
  ntagCardPageWrite нет проверки записи на чип путем считывания и сравнения
  данных из буфера, что существенно снижает надежность отметки. Может стоило бы
  сохранить сей полезной момент в новой версии?

Phylosof, 2020-03-09:
Благодарю, я уже нашел) В том и дело что аналогичная функция в третьей версии не верифицирует запись данных.

sembruk, 2020-03-10:
@mars2860 , эти изменения произошли в версии 2.6? Была ли для этого какая-то особая причина?

mars2860, 2020-03-10:
Да, в версии 2.6 я не стал делать проверку отметки после записи. Дополнительная операция чтения - увеличивает время работы базы с чипом. К тому же посчитал это излишним, так как чип MFRC522 сам отлавливает сбои при чтении/записи и если будет сбой при записи отметки, то MFRC522 об этом сообщит и выкинет в итоге false

closed 2020-03-18 by Phylosof