sportiduino/sportiduino #91: Криптозащита отметок в чипе

alexandervolikov, 2020-09-16:
Навеяно обсуждением в чате одного крупного московского соревнования.

Раньше можно было проткнуть карточку иголкой, эмулируя работу компостера, теперь можно записать отметку в чипе с помощью телефона с NFC и навыков программирования.

Защититься от этого можно, считав дампы со станций после проведения соревнования и сравнить с результатами. Но, обычно, станции снимаются позже, выявить подлог можно сильно позже, ну и дополнительные действия для этого нужно делать.

Как решение, можно вставить хэш от времени и пароля станции в отметку, например, SHA-256.
Сейчас структура отметки в 4 байта: 8бит-номер станции, 24бит - остаток от unixtime. 24бита времени позволяют проводить соревнования в течение 194 дней после инициализации чипа, что, может быть, и несколько излишне. Вполне можно уменьшить этот срок в 2-8 раз, высвободив немного памяти для хэша. Например, сделать так:
8бит-номер станции, 2бит - хэш, 22 бит - остаток от unixtime. 

Как вам идея?

halny123, 2020-09-16:
Идея замечательная, т.к. проблема рано или поздно все равно встанет. 
Перед забегом очистка чипа - хорошая практика, а месяца должно хватить для проведения мероприятия по любому виду соревнований.
Считывающая станция должна проверять корректность хэша и сигнализировать какой нибудь "свирестелью" в случае обнаружения подделки, не возлагая эту задачу на ПО обработки.

sembruk, 2020-09-16:
Идея хорошая. Нужно проверить, сколько времени займёт вычисление хэша в момент отметки. Если будет долго, можно при инициализации станции один раз вычислить хэш от номера и пароля и держать его в оперативной памяти.

alexandervolikov, 2020-09-16:
Проверил, получается 22 мс. Сложно сказать на сколько критично замедление на эту величину. Но я ещё поизучаю матчасть, и возможности для оптимизации библиотеки.

Использовал вот эту библиотечку: https://github.com/manutenfruits/Arduino-SHA-256
По такому примеру:
```
#include "sha256.h"

void setup() {
  
  Serial.begin(9600);
  uint8_t temp_hash = 0;
  Sha256.init();
  
  Serial.println(millis());
  for (uint32_t i = 0; i<100; i++){
      
    Sha256.print(i);
    temp_hash = Sha256.result()[0];

  }
  
  Serial.println(millis());
  
}

void loop() {
  
}
```

Pavel-Y, 2020-09-16:
Использовать 2 бита от SHA-256 довольно расточительно, если функция долгая. При таком множестве возможных значений верификаторов (аж 4 штуки) можно взять любой самый быстрый хэш.

Pavel-Y, 2020-09-16:
Еще вариант - писать в фиксированное место на чипе что-то вроде XOR(hash(mark1), ..., hash(markN). Тогда этот блок в N=30..50 раз быстрее будет изнашиваться. Не знаю, критично это или нет.
Достоинство - возможна обратная совместимость: станции со старой прошивкой будут "фиксированное место" игнорировать (в протоколе есть незанятое место).

Pavel-Y, 2020-09-16:
> Идея хорошая. Нужно проверить, сколько времени займёт вычисление хэша в момент отметки. Если будет долго, можно при инициализации станции один раз вычислить хэш от номера и пароля и держать его в оперативной памяти.

Если "хэш" не зависит от записываемой на чип информации, взлом протокола = написать чуть больше кода, чем для просто записи фейковых отметок на чип. 

Сценарий: чип1 ходит по станциям 1..N/2, чип2 от N/2 до N, потом они встречаются и имеют всю информацию для записи произвольных отметок на оба чипа.

Pavel-Y, 2020-09-16:
Из предыдущего комментария следует, что хэш надо вычислять в том числе от UID'a чипа. Хотя если UID перезаписываемый, тоже защита = "злоум-ку писать еще чуть больше кода".

Pavel-Y, 2020-09-16:
Вообще, SFR до сих пор как-то живет безо всякой защиты для перенастройки станций и перезаписи чипов. Видимо, за счёт закрытого протокола и лени*доброты заинтересованных лиц :)

alexandervolikov, 2020-09-16:
Попробовал SHA-1, считает уже 11 мс. В принципе, на мой взгляд, уже не критично для работы системы. Но может, действительно, попробовать поискать что-то менее мощное раз нам только несколько битов нужно.

Ну или, как предложили в телеграмме, вспомнить про защиту от записи, встроенную в чипе.

ЗЫ. Вот тут более актуальная библиотека, с лицензией:  https://github.com/daknuett/cryptosuite2

alexandervolikov, 2020-09-17:
Протестировал защиту, встроенную в чипы Ntag. Работает, настраивается довольно легко и гибко, вполне надежно. Пароль с чипа не выдрать, можно поставить защиту от брут-форса.
На скорость чтения чипа установка пароля не повлияла (я ставил пароль только на запись), а вот скорость записи, закономерно, возросла с 30 до 40 мс на страницу. То есть, разница во все те же 10 мс на отметку. 
В чипы Mifare не лез, но подозреваю, что там примерно тоже самое должно получаться.

Плюс встроенной защиты вижу в том, что довольно легко переделать систему без существенного изменения протоколов записи отметки.  Функция авторизации по паролю реализована в библиотеке RC522, существенной доработки кода не требуется. Минусы вижу в усложнении всей процедуры действий с чипами и станциями, жесткой привязке к паролям. Нужно будет следить за чипами и менять пароли на них вместе со сменой паролей на станциях. Либо, например, очищать чипы от пароля при считывании результатов. Продумать всякие варианты по подобию завалявшихся у кого-то после прошлогодних соревнований чипов или станций со старыми паролями.

В целом, склоняюсь к тому, что правильней реализовать встроенную в чипы защиту.

Pavel-Y, 2020-09-17:
> Протестировал защиту, встроенную в чипы Ntag. Работает, настраивается довольно легко и гибко, вполне надежно. Пароль с чипа не выдрать, можно поставить защиту от брут-форса.
> На скорость чтения чипа [...]

Чипы Mifare Classic всегда используют защиту паролем. Известен пароль по умолчанию, с которым они приходят с фабрики.

vtararin, 2020-11-14:
Может проще писать время и метку станции шифрованные через 3DES паролем станции и номером чипа. Просто, быстро, на другой чип не перенести.

alexandervolikov, 2020-11-14:
> 
> 
> Может проще писать время и метку станции шифрованные через 3DES паролем станции и номером чипа. Просто, быстро, на другой чип не перенести.

В этом случае теряется часть возможной функциональности - чтобы получить данные с чипа нужен пароль станции. Соответственно, участник сам уже не сможет считать свой чип. В общем-то, может оно и не надо, но зачем урезать функционал.

Pavel-Y, 2020-11-14:
> В этом случае теряется часть возможной функциональности - чтобы получить данные с чипа нужен пароль станции. Соответственно, участник сам уже не сможет считать свой чип. В общем-то, может оно и не [...]

А мне идея понравилась. Не вижу юз кейза, когда участнику нужно было бы считать свой чип - своим телефоном?
Все судьи пароль станции знают. Можно исползовать пароль по умолчанию.

Другое дело, что если в чипы уже встроена защита, то логично использовать ее (и не смешить людей заодно).