sportiduino/sportiduino #92: Разделить мастер-чипы установки пароля и (режимы и усиления)

Pavel-Y, 2020-09-17:
Речь про чип с байтом == 254.
Когда пользователь хочет поменять только пароль станции, ему не нужно ничего знать про "режимы" и "усиление". 
Аналогично при перенастройке усиления станции пользователь не обязательно хочет менять ее пароль.
Поэтому логично было бы сделать два разных чипа: один для переустановки пароля станции, второй для "режимов" и "усиления". Например, второй - с байтом == 248.

В качестве полумеры можно сделать дефолтные значения для режима и усиления, которые не меняют их в станции, но это противоречит принципу "один объект - одна задача".

Phylosof, 2020-09-17:
Так они и так разделены. Речь идет про программу SportidunoPQ?

Pavel-Y, 2020-09-17:
> Так они и так разделены. Речь идет про работу в программе SportidunoPQ?

Сейчас на странице https://github.com/sportiduino/sportiduino/blob/master/Doc/ru/Card.md они НЕ разделены.
Речь о новой разработке.

Phylosof, 2020-09-17:
Действительно, в описании это один чип. Надо глянуть, как это сейчас реализовано в прошивке, так как в интерфейсе SportiduinoPQ они разделены.

closed 2025-01-27 by sembruk