sportiduino/sportiduino #97: Чип с возможностью бесконтактной/контактной отметки.

mforis, 2020-10-20:
Добавить в станцию отметки модуль NRF24L01, а чип изготовить на микроконтроллере NRF52832. Стоимость микроконтроллера 1$. Примерная стоимость изготовления чипа 2-3$, что в 20-30 раз дешевле чипа SportIdent Air+. Буду разрабатывать печатную плату. Необходима помощь в написании протокола и прошивки.

MatyasovM, 2020-11-09:
Ты не думал о том, чтобы не делать собственный чип, а спроектировать поддержку sport ident чипов. То есть сделать станцию, которая бы могла читать чипы, которые все используют уже ?

mforis, 2020-11-09:
@MatyasovM , хм, нужно знать тогда как работает SportIdent Air+, а исходников софта и внутренностей чипа вряд-ли кто-то даст. Поэтому, во всю работаем с разработкой бесконтактно-контактного чипа.

MatyasovM, 2020-11-09:
Это, понятно, просто если ты хочешь чтобы проект вылился во то-то большее чем просто хобби, то нужно подумать о том, кто будет покупать такие чипы и станции. А поскольку есть проверенный, надежный sportident, то есть сомнения в том будут ли на соревнованиях (на больших) использовать новую систему. Я сейчас сам пытаюсь построить сделать станцию, которая могла бы удаленно отметку отправлять, при чем без использования GSM (Возможно с помощью LoRa).
На данный момент думаю, как их протокол взломать (RFID на 125khz и 13mhz не работают). В документации в чипам написано, что онии используют SRR на частоте 2.4ghz. Я уже видел репозиторий, где человек пытался что-то с NRF сделать, вот и подумал может у тебя есть какие то соображения на этот счет ?

mforis, 2020-11-09:
@MatyasovM ,

MatyasovM, 2020-11-09:
Ага, спасибо. Ты кстати один всем занимаешься или уже команда есть ?

mforis, 2020-11-09:
@MatyasovM , давай в телегу лучше, чтобы гит не загрязнять https://t.me/max_mirnyi