sportiduino/sportiduino #98: Разница между кодом SportIduinoPQ и протоколом в GitHub'e

sportiduino-lite, 2020-10-31:
Код SportIduinoPQ, сейчас реализует не протокол, выложенный в этом же репозитории - https://github.com/sportiduino/sportiduino/blob/master/Doc/ru/Card.md , а какой-то другой. Он не описан в явном виде, только в коде SportIduinoPQ.

Тем, кто пишет код не для SportIduinoPQ, а для других программ, нужен чётко описанный логичный протокол.

sportiduino-lite, 2020-10-31:
Например, вот этот код 
`    def init_time_card(self, time=datetime.today()):
        """Initialize card for writing time to base station.
        @param time: Time for base station (default current time).
        """
        params = b''
        params += int2byte(time.year - 2000)
        params += int2byte(time.month)
        params += int2byte(time.day)
        params += int2byte(time.hour)
        params += int2byte(time.minute)
        params += int2byte(time.second)`

скорее всего не соответствует протоколу "месяц год-2000 день 0 час минута секунда 0".

А также запись 0 для MifareClassic1k чипа участника в байт [2], где для NTag 0 означает, что чип очищен, а для участников там 3, 5 или 6, создает неконсистентный протокол. Нужно исправить на ненулевое фиксированное значение, например на 8 (используется в другой программе).

sembruk, 2020-11-05:
Значение 0 записывается не только для MifareClassic1k, но и для всех неNTag чипов.
Этот байт используется только в прошивках старее версии 6. Он оставлен для совместимости, и может быть убран в любой момент (кроме значения 255 для мастер-чипа). Использовать его значение о типе чипа и тем более как индикатор очистки чипа в новом софте нельзя!

sembruk, 2020-11-05:
> Код SportIduinoPQ сейчас реализует не протокол, выложенный в этом же репозитории - https://github.com/sportiduino/sportiduino/blob/master/Doc/ru/Card.md , а какой-то другой. Он не описан в явном виде, только в коде SportIduinoPQ.

С одним частным случаем разобрались. Теперь прошу указать на другие несоответсвия описания протокола, и реального поведения.

closed 2022-01-28 by sembruk