sportiduino/sportiduino #105: Не верная дата при работе через SportIden reader

vtararin, 2021-03-28:
При чтении чипов  через SportIden reader дата тметки всегда 2000-01-01

sembruk, 2021-03-29:
Какие версии SportIden Reader, SportiduinoPQ и прошивки читающей станции?
Если можно, прошу приложить скриншоты чтения чипов в SportiduinoPQ и в Reader.

vtararin, 2021-03-29:
sportiden reader v. 3.2 14.10.2020 
[event.csv.log](https://github.com/sportiduino/sportiduino/files/6223690/event.csv.log)
станция сопряжения - master
[image]

sembruk, 2021-04-03:
Подосовывание даты в ответ на get_time (0xF7) не помогло, а в формате отметки, записанной на SI Card, есть только время и день недели, даты нет. Если SportIdent Reader правильно читает дату с настоящего чипа, то пока непонятно, как он это делает.

closed 2025-01-27 by sembruk