sportiduino/sportiduino #110: Здравствуйте! Научите,пожалуйста, "прошивать".

mrKamanov, 2022-10-10:
Здравствуйте! Научите,пожалуйста, "прошивать".
Здравствуйте, меня зовут Сергей. Пытаюсь повторить проект для детей в школе. Начал проект со сборки базовой станции- спаял.
Но теперь необходимо ее прошить(так написано в инструкции на этом сайте). Скачал версию arduino ide 1.8.7 как на скриншоте в статье, указал пусть к папкам и скопировал файлы hex как написано, внес изменения в файл board.... Но ничего не появляется? нет в списке платы sportiduino( работа встала,не понима, что делать). Прошу помощи. Спасибо.

п.с. У меня есть ардуино нано и програматор, но ими ни разу не пользовался.

sembruk, 2022-10-12:
Проверьте файл board.

mrKamanov, 2022-10-12:
Здравствуйте, В конце файла вставил:
##############################################################

sportiduino.name=Sportiduino

sportiduino.build.mcu=atmega328p
sportiduino.build.f_cpu=8000000L
sportiduino.build.board=AVR_SPORTIDUINO
sportiduino.build.core=arduino
sportiduino.build.variant=standard

sportiduino.upload.tool=avrdude
sportiduino.upload.protocol=arduino
sportiduino.upload.maximum_size=32256
sportiduino.upload.maximum_data_size=2048
sportiduino.upload.speed=38400

sportiduino.bootloader.tool=avrdude
sportiduino.bootloader.low_fuses=0xE2
sportiduino.bootloader.high_fuses=0xDE
sportiduino.bootloader.extended_fuses=0xFF
sportiduino.bootloader.unlock_bits=0x3F
sportiduino.bootloader.lock_bits=0x0F

menu.bootloader=Bootloader

sportiduino.menu.bootloader.optiboot=Optiboot 38400
sportiduino.menu.bootloader.optiboot.bootloader.file=optiboot/optiboot8_38400_sportiduino.hex

sportiduino.menu.bootloader.optiboot_led=Optiboot 38400 with LED
sportiduino.menu.bootloader.optiboot_led.bootloader.file=optiboot/optiboot8_38400_sportiduino_led.hex

############################################################## 

nano.name=Arduino Nano

nano.upload.tool=avrdude
nano.upload.protocol=arduino

nano.bootloader.tool=avrdude
nano.bootloader.unlock_bits=0x3F
nano.bootloader.lock_bits=0x0F

nano.build.f_cpu=16000000L
nano.build.board=AVR_NANO
nano.build.core=arduino
nano.build.variant=eightanaloginputs

## Arduino Nano w/ ATmega328P (without reset)
## --------------------------
nano.menu.cpu.atmega328noreset=ATmega328P (without reset)

nano.menu.cpu.atmega328noreset.upload.maximum_size=30720
nano.menu.cpu.atmega328noreset.upload.maximum_data_size=2048
nano.menu.cpu.atmega328noreset.upload.speed=38400

nano.menu.cpu.atmega328noreset.bootloader.low_fuses=0xFF
nano.menu.cpu.atmega328noreset.bootloader.high_fuses=0xDA
nano.menu.cpu.atmega328noreset.bootloader.extended_fuses=0xFD
nano.menu.cpu.atmega328noreset.bootloader.file=optiboot/optiboot8_atmega328_38400_without_reset.hex

nano.menu.cpu.atmega328noreset.build.mcu=atmega328p

##############################################################

mrKamanov, 2022-10-12:
после многократной переустановки sportiduino и ATmega328P (без сброса) появились, видимо, проблема с кэшом была какая-то. Вечером попробую загрузить скетч. Спасибо вам за ответ.

mrKamanov, 2022-10-12:
теперь очередная сложность:
при загрузке скетча "выяснилось" что чип на ардуино нано atmel mega 168. ( переставил процессор на ATmega 168- скетч установился, но с надписью "Недостаточно памяти, программа может работать нестабильно.") Скачал программу SportiduinoPQ , запустил, при подключении пишет 
"2022-10-12 17:44:54
Подключена станция сопряжения v1.9.0 на порту COM3"

При попытке записать чип: 
"Начинаю подготовку чипа участника
Ошибка: Чип не обнаружен"
( у меня их штук 10 белые карточки, 10 синие брелки, и ни один не видит).

Скачал приложение на телефон NFC Tools - оно видит эти чипы , записывает их и читает.
Скачал приложение Sportiduino Lite на телефон -  при попытке записать чип пишет "Ошибка: Transceive failed"

Что я делаю не так? в чем может быть проблема?

sembruk, 2022-10-12:
1. Стоит проверить правильность подключения платы RC522 к Arduino Nano.
2. На Atmega168 никто пока эти прошивки не проверял.
3. Sportiduino Lite не надо использовать. Это стороннее приложение, актуальные версии прошивок в нём не поддерживаются. Вот правильное приложение — https://github.com/sportiduino/sportiduinoapp

mrKamanov, 2022-10-12:
Не помогло, контакты все перепроверил. 
Подключена станция сопряжения v1.9.0 на порту COM3

Начинаю читать чип
Ошибка: Нет ответа

Начинаю подготовку чипа участника
Ошибка: Чип не обнаружен

попробовал несколько других rfid-rc522- чип не видят.

попробую поискать ардуино нано с чипом Atmega328

mrKamanov, 2022-10-15:
Да, видимо дело было именно в чипе. Станция заработала чипы теперь видит, светодиод мигает, но нет звукового сигнала. В чем может быть проблема?

sembruk, 2022-10-15:
> Да, видимо дело было именно в чипе.

Может быть, на плате Nano с Atmega168 выводы расположены по-другому?

> нет звукового сигнала. 

Можно прозвонить цепь до буззера и проверить его активное сопротивление.

mrKamanov, 2022-10-15:
А вы не подскажите как это осуществить, моих знаний не хватает.

mrKamanov, 2022-10-15:
я имею ввиду какое сопротивление выставить нужно на мультиметре

mrKamanov, 2022-10-15:
поставил на 20к показывает 0.02

sembruk, 2022-10-15:
Сопротивление в норме. Ещё можно проверить пайку резистора на 47 Ом.

mrKamanov, 2022-10-15:
Спасибо за ответ

mrKamanov, 2022-10-15:
Да, помогло, перепаял резистор, еще раз спасибо!!!

closed 2025-01-27 by sembruk